qmk/qmk_firmware #24557: Connect USB cable to the right hand side, DIP switches will be disabled.

s-show, 2024-11-02:
### Issue Description

I have a problem with the split keyboard I designed, where the DIP switch stops working when the USB cable is connected to the right hand side instead of the left hand side.

There is one DIP switch attached to the left hand side, and when the USB cable is connected to the left hand side, it works fine. However, when the USB cable is connected to the right hand side, the DIP switch stops working, even though the typing and OLED display are the same as when the cable is connected to the left hand side.

So, for debugging purposes, I added the following code to `keymap.c`.

```c
// keymap.c
bool dip_switch_update_mask_user(uint32_t state) {
    #ifdef CONSOLE_ENABLE
uprintf(“switch state: %lu\n”, state);
#endif
(Processing according to the DIP switch state continues)
```

Then, connect the USB cable to the left side and switch the DIP switch on and off, `switch state: 1` or `switch state: 0` will be displayed in the HID Console of QMK Toolbox. However, connect the USB cable to the right side, nothing will be displayed in the HID Console even if switch the DIP switch on and off.

The keyboard is running on Waveshare's RP2040-Zero, and left/right judgment is achieved by setting `GP13` to `split.handedness.pin`.

The settings for the parts that are thought to be related to this problem are as follows, but is there a way to solve this problem?

`keyboard.json`
```json
    "dip_switch": {
        "enabled": true,
        "pins": [ "GP12", "GP11", "GP10", "GP9" ]
    },
    "split": {
        "enabled": true,
        "matrix_pins": {
            "right": {
                "cols": ["GP14", "GP15", "GP26", "GP27", "GP28", "GP29"],
                "rows": ["GP0", "GP1", "GP2", "GP3"]
            }
        },
        "handedness": {
            "pin": "GP13"
        },
        "serial": {
            "driver": "vendor",
            "pin": "GP8"
        }
    },
```

```c
// config.h
#define SERIAL_USART_HALF_DUPLEX
#define SPLIT_OLED_ENABLE
#undef SPLIT_USB_DETECT
```

[image]

sigprof, 2024-11-03:
Currently the DIP Switch feature does not fully support split boards — only the master side actually polls the DIP switches connected to that side (the `dip_switch_task()` call in `quantum_task()` is after the `if (!is_keyboard_master()) return;` check).  The only exception is if you are using `DIP_SWITCH_MATRIX_GRID` — in that case the master side will see the state of DIP switches integrated into the key matrix, even if some of those switches are on another side.

Adding support for your use case would require some new code to handle the DIP switch state transfer to the master side, and probably some new ways to configure the feature (similar to what had been done for encoders, where you can have a different number of encoders on each side).  The existing support for `DIP_SWITCH_PINS_RIGHT` assumes that both sides have the same number of DIP switches, but only the ones which are on the master side would actually be used.

s-show, 2024-11-04:
Thank you for your reply.
I think I will solve this problem using `DIP_SWITCH_MATRIX_GRID`.
Also, before designing this keyboard, I tried to control the DIP switches on the right side by setting `DIP_SWITCH_PINS_RIGHT`, but I was unable to control the DIP switches on the right side. I now understand why.
Thank you for your advice.

closed 2024-11-04 by s-show