qmk/qmk_firmware #24576: [Feature Request] Hybrid HID Descriptor/NKRO on by default

SpoonyAllen, 2024-11-07:
### Description

Gave QMK configurator a whirl using a rp2040 pico and the results are incredible. I want to first and foremost thank everyone involved with the project.

I did come across a situation however where my finished project would either not be detected by my computers bios and would not automatically run following OS startup. If i disconnect and reconnect the device after OS startup, it works perfect. It seems that (at least the rp2040) code does not seem have a hybrid HID descriptor.

 Is there any possible chance to add this?

fauxpark, 2024-11-07:
Your issue has nothing to do with QMK having or not having a "hybrid HID descriptor", whatever that is.

drashna, 2024-12-01:
If it's a split keyboard, defining `SPLIT_WATCHDOG_ENABLE` may help.

closed 2025-04-01 by github-actions[bot]