qmk/qmk_firmware #24591: Halves of split keyboard not seeing each other

Kineolyan, 2024-11-11:
### Issue Description

As many before, I am having trouble with my custom split keyboard.

It is made of two halves, both using a Teensy 2.0. I am using a matrix pin on the left half to define the handedness. For a reason I don't remember, I am not using the same pins for both sides.

I am aware of [the split guide](https://docs.qmk.fm/features/split_keyboard) and [info.json ref](https://docs.qmk.fm/reference_info_json#split-keyboard) and still have issues.

I have a setup that is working like a charm [here](https://github.com/Kineolyan/qmk_firmware/tree/main/keyboards/handwired/splx/full) but it is based on an old version of QMK (0.23.3).
I have been reading the migration guides to upgrade to the latest and got each half working separately.
But they are still not talking with each other.

I tested many different serial pins - currently D3 - as well as setting this in `keyboard.json` or the `config.h`. But it still is not working.
My current work is on [this branch](https://github.com/Kineolyan/qmk_firmware/tree/upgrade-kernel/keyboards/handwired/splx/full).

Without changing any hardware wiring, I can flash a not-working version of the system from my upgrade branch and repair everything by switching back to my main branch.
This tells me that the wiring is correct, that boards are correctly powering each other. When working, I can plug any half to my computer and have the keyboard working.
It seems that there is a new config for the serial driver, which may be what I need. I am currently using serial + bitbang, to use a single wire for the whole setup, to avoid resistors.

After many research, I haven't found any way to add some debug messages and check if the serial is somehow complaining. So I am turning to you, dear maintainers.
_If this kind of message is better suited for another channel, please forgive me and redirect me._

zvecr, 2024-11-11:
https://github.com/Kineolyan/qmk_firmware/blob/main/keyboards/handwired/splx/full/info.json#L54

vs

https://github.com/Kineolyan/qmk_firmware/blob/upgrade-kernel/keyboards/handwired/splx/full/keyboard.json#L55

Teensy do not have working vbus detection, so this setting is required.

Kineolyan, 2024-11-11:
Thanks for spotting this difference. A bit disappointed to not have seen it before :disappointed: 
I will test this asap.
After reading again and again this section https://docs.qmk.fm/features/split_keyboard#teensy-boards, I am still struggling a bit to understand, maybe due to a lack of understand of what VUSB is meant to be.
But as I was writing my questions to you, I eventually got my answers. :duck: 
USB detect will allow to find which board should be the master. Many boards automatically detect this with VUSB, which easily tells if the power comes from USB or another source - like the other half.
And there is different from handedness, which only tells which side is left and which is right.
_(please, confirm my understanding if you have time)_

Again, testing this asap and coming back at you to close the issue :crossed_fingers: or with more questions :shrug:

Kineolyan, 2024-11-11:
This Was It :confetti_ball: 
Many thanks to you @zvecr .
Closing this issue as it is resolved

closed 2024-11-11 by Kineolyan