qmk/qmk_firmware #24599: [Bug] Mouse key MS_BTN1 ~ MS_BTN8 are not working when PS/2 trackpoint enabled with RP2040 MCU

tachikomachann, 2024-11-14:
### Describe the Bug

Hi, I'm trying to create a keyboard with trackpoint module for myself. 
I've got the PS/2 trackpoint module and use Raspberry PI PICO (RP2040 MCU) to test my QMK firmware.
I've enabled the PS/2 mouse and vendor driver in keyboard.json and the trackpoint works fine. Then I tried to update my keymap to support sending mouse button click(left/middle/right button click) from my keyboard, but I find it is not working. The tricky thing is only MS_BTN1 ~ MS_BTN8 are not working, others works fine(e.g. the wheel scroll keycode MS_WHLU and MS_WHLD). 
I tried to disable the PS/2 module and driver then the result is MS_BTN1 ~ MS_BTN3 are working again. I can send mouse click from my keyboard.
is this a conflict between RP2040 ps2 vendor driver and mouse key feature? How can I resolve this and let my trackpoint and mouse keycode work together?

I used the latest master branch qmk_firmware repository.
Below are my related files (my keyboard name is nomouse):
rule.mk:
```shell
# Bootloader selection
BOOTLOADER = rp2040
MOUSEKEY_ENABLE = yes
BOARD = GENERIC_RP_RP2040
CONSOLE_ENABLE = yes
```
nomouse.h:
```C
#pragma once
#include "quantum.h"

```
keyboard.json:
```json
{
    "manufacturer": "Tachikoma",
    "keyboard_name": "Nomouse",
    "maintainer": "Tachikoma",
    "bootloader": "rp2040",
    "diode_direction": "COL2ROW",
    "features": {
        "bootmagic": true,
        "command": true,
        "console": true,
        "extrakey": true,
        "mousekey": true,
        "nkro": true
    },
    "matrix_pins": {
        "cols": [ "GP2", "GP3", "GP4", "GP5", "GP6", "GP7", "GP8", "GP9", "GP10", "GP11", "GP12", "GP13", "GP14", "GP15" ],
        "rows": [ "GP16", "GP17", "GP18", "GP19", "GP20"]
    },
    "processor": "RP2040",
    "url": "",
    "usb": {
        "device_version": "1.0.0",
        "pid": "0x0000",
        "vid": "0xFEED"
    },
    "ps2": {
        "clock_pin": "GP1",
        "data_pin": "GP0",
        "driver": "vendor",
        "enabled": true,
        "mouse_enabled": true
    },
    "mouse_key": {
        "enabled":true
    },
    "layouts": {
        "LAYOUT": {
            "layout": [
                {"matrix": [0, 0], "x": 0, "y": 0},
                {"matrix": [0, 1], "x": 1, "y": 0},
                {"matrix": [0, 2], "x": 2, "y": 0},
                {"matrix": [0, 3], "x": 3, "y": 0},
                {"matrix": [0, 4], "x": 4, "y": 0},
                {"matrix": [0, 5], "x": 5, "y": 0},
                {"matrix": [0, 6], "x": 6, "y": 0},
                {"matrix": [0, 7], "x": 7, "y": 0},
                {"matrix": [0, 8], "x": 8, "y": 0},
                {"matrix": [0, 9], "x": 9, "y": 0},
                {"matrix": [0, 10], "x": 10, "y": 0},
                {"matrix": [0, 11], "x": 11, "y": 0},
                {"matrix": [0, 12], "x": 12, "y": 0},
                {"matrix": [0, 13], "x": 13, "y": 0, "w": 2},

                {"matrix": [1, 0], "x": 0, "y": 1, "w": 1.5},
                {"matrix": [1, 1], "x": 1.5, "y": 1},
                {"matrix": [1, 2], "x": 2.5, "y": 1},
                {"matrix": [1, 3], "x": 3.5, "y": 1},
                {"matrix": [1, 4], "x": 4.5, "y": 1},
                {"matrix": [1, 5], "x": 5.5, "y": 1},
                {"matrix": [1, 6], "x": 6.5, "y": 1},
                {"matrix": [1, 7], "x": 7.5, "y": 1},
                {"matrix": [1, 8], "x": 8.5, "y": 1},
                {"matrix": [1, 9], "x": 9.5, "y": 1},
                {"matrix": [1, 10], "x": 10.5, "y": 1},
                {"matrix": [1, 11], "x": 11.5, "y": 1},
                {"matrix": [1, 12], "x": 12.5, "y": 1},
                {"matrix": [1, 13], "x": 13.5, "y": 1, "w": 1.5},

                {"matrix": [2, 0], "x": 0, "y": 2, "w": 1.75},
                {"matrix": [2, 2], "x": 1.75, "y": 2},
                {"matrix": [2, 3], "x": 2.75, "y": 2},
                {"matrix": [2, 4], "x": 3.75, "y": 2},
                {"matrix": [2, 5], "x": 4.75, "y": 2},
                {"matrix": [2, 6], "x": 5.75, "y": 2},
                {"matrix": [2, 7], "x": 6.75, "y": 2},
                {"matrix": [2, 8], "x": 7.75, "y": 2},
                {"matrix": [2, 9], "x": 8.75, "y": 2},
                {"matrix": [2, 10], "x": 9.75, "y": 2},
                {"matrix": [2, 11], "x": 10.75, "y": 2},
                {"matrix": [2, 12], "x": 11.75, "y": 2},
                {"matrix": [2, 13], "x": 12.75, "y": 2, "w": 2.25},

                {"matrix": [3, 1], "x": 0, "y": 3, "w": 2.25},
                {"matrix": [3, 2], "x": 2.25, "y": 3},
                {"matrix": [3, 3], "x": 3.25, "y": 3},
                {"matrix": [3, 4], "x": 4.25, "y": 3},
                {"matrix": [3, 5], "x": 5.25, "y": 3},
                {"matrix": [3, 6], "x": 6.25, "y": 3},
                {"matrix": [3, 7], "x": 7.25, "y": 3},
                {"matrix": [3, 8], "x": 8.25, "y": 3},
                {"matrix": [3, 9], "x": 9.25, "y": 3},
                {"matrix": [3, 10], "x": 10.25, "y": 3},
                {"matrix": [3, 11], "x": 11.25, "y": 3},
                {"matrix": [3, 12], "x": 12.25, "y": 3, "w": 1.75},
                {"matrix": [3, 13], "x": 14, "y": 3, "w": 1},

                {"matrix": [4, 0], "x": 0, "y": 4, "w": 1.25},
                {"matrix": [4, 2], "x": 1.25, "y": 4, "w": 1.5},
                {"matrix": [4, 3], "x": 2.75, "y": 4, "w": 1.25},
                {"matrix": [4, 5], "x": 4, "y": 4, "w": 2},
                {"matrix": [4, 8], "x": 8, "y": 4, "w": 2},
                {"matrix": [4, 10], "x": 10, "y": 4, "w": 1.5},
                {"matrix": [4, 11], "x": 11.5, "y": 4, "w": 1.5},
                {"matrix": [4, 12], "x": 13, "y": 4, "w": 1.5},
                {"matrix": [4, 13], "x": 14, "y": 4, "w": 1}
            ]
        }
    }
}
```

config.h:
```
#pragma once

```

keymap.c:
```C
// Copyright 2023 QMK
// SPDX-License-Identifier: GPL-2.0-or-later

#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

    [0] = LAYOUT( /* 0: qwerty */
        KC_ESC,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSLS,
        // KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_BSPC,
        // this row for mouse key testing
        KC_1,  MS_BTN1,    MS_BTN2,    MS_BTN3,    MS_WHLU,    MS_WHLD,    MS_WHLL,    MS_WHLR,    KC_MS_BTN2,    MS_BTN1,    KC_P,    KC_LBRC, KC_RBRC, KC_BSPC,
        KC_LCTL,          KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT, KC_ENT,
                 KC_LSFT, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT, KC_HOME,
        KC_LCTL,          KC_LGUI, KC_LALT,          KC_SPC,                    KC_ENT,           KC_RALT, KC_RGUI, KC_RCTL, KC_END
    ),
};

```

My config refers to these docs:
https://docs.qmk.fm/features/mouse_keys
https://docs.qmk.fm/features/ps2_mouse#rp2040-pio-version

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

gamelaster, 2024-11-14:
Hi, from PS/2 mouse, there is bitfield which just contains left, middle and right button state. It is passed then to QMK PS/2 interface. I am just using the keyboard with PS/2 trackpoint and mouse buttons works just fine.

I have no idea what `mouse_key` property does at all, so I can't help with this. Although, I don't think the issue is in RP2040 PS/2 PIO driver.

tachikomachann, 2024-11-16:
> Hi, from PS/2 mouse, there is bitfield which just contains left, middle and right button state. It is passed then to QMK PS/2 interface. I am just using the keyboard with PS/2 trackpoint and mouse [...]

do you know how can I debug this? I dont know how to resolve it now... 😂 @gamelaster

gamelaster, 2024-11-16:
@tachikomachann I am sorry, no clues :(

tachikomachann, 2024-11-17:
@gamelaster I do more test and try to dig more details:
1. Add debug log in `process_record_user` and find the MS_BTN1 ~ MS_BTN3 keycode can be processed in this, so I think the pin connection and key press event should be correct?  
[image]

3. I keep the the ps/2 mouse config and just remove ps/2 trackpoint module from my breadboard. Mouse key can send left/middle/right click after removing the module.
4. I find the MS_BTN3 become the scroll button(i.e. **hold the MS_BTN3, trackpoint movement will become mouse wheel scroll** ) even though I don't enable any scroll button setting.

According above details, maybe I can confirm that the mouse key buttons behavior is overridden by ps/2 module? Can I override the behavior of MS_BTN1 ~ MS_BTN3 in `process_record_user` as a workaround?  I'm not familiar with ps/2 driver code and don't know how to troubleshoot in ps/2 code

HattDroid, 2025-06-15:
I'm facing the exact same issue. Thanks for reporting it.

@tachikomachann you don't have to physically remove the PS/2 module, turning off the mouse property of the PS2 block is enough to make the mouse buttons work:

`"ps2": {
     "mouse_enabled": false`

What I don't understand is that the 0x00D2 and the other mouse buttons keycodes are still sent and received when enabling the PS/2 module, it just looks like the non-existing keys of the PS/2 module are taking precedence over the mouse button keycodes sent.

Would this work by maybe giving an option to disable the PS/2 module keys?
Please bear in mind that I haven't looked at the code of the PS/2 driver, it's only a suggestion based on how the board and system behave.

How can we get this to be fixed? Would anyone have pointers maybe?

HattDroid, 2025-06-17:
@tachikomachann After testing around a few settings related to PS/2, this is the setting that finally allowed mouse clicks from the keyboards button when using a PS/2 trackpoint without buttons:

`#define PS2_MOUSE_USE_REMOTE_MODE
`

The only problem I have now is switching to remote mode instead of stream mode is extremely laggy for most of the time which makes typing impossible.

Would anyone have an explanation of why remote mode actually works for mouse clicks now? And how would it be possible to get those clicks to work in stream mode?

vzsky, 2025-08-04:
My bad