qmk/qmk_firmware #24606: [Bug] SPI used pins can double used on keyboard.json.

DreaM117er, 2024-11-15:
### Describe the Bug

[globsck.zip](https://github.com/user-attachments/files/17770069/globsck.zip)

The code package is here.

This is a mouse driven by PMW3389 sensor, and I test on bread board, accidently found this bug.
- double used GP3 and GP4, and can compile from QMK_MSYS.

[image]

### Is AutoHotKey / Karabiner installed

- [X] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2024-11-15:
Not sure what the issue is, nor what you’re expecting behaviour-wise. Are you saying that the same pins can be configured for two things?

drashna, 2024-12-01:
Do you mean that it is not checking for overlapping usage of pins?

If so, that's expected, as there isn't anything checking for overlapping usage.

closed 2025-01-11 by tzarc