qmk/qmk_firmware #24613: [Bug]  Makefile 407.

agsb, 2024-11-17:
### Describe the Bug

Hello,

I just clone the qmk, and try make ferris/sweep:default then got a:

Makefile:407: *** missing separator (did you mean TAB instead of 8 spaces?).  Stop.

Any Tip ?

And, yes i had checked and there is a TAB at line 407.

thanks

### Keyboard Used

ferris/sweep:default

### Operating System

WSL

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/agsb/Gits/qmk_firmware
Ψ Detected Linux (WSL, Ubuntu 24.04.1 LTS).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.26.11
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 7.3.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2024-12-01:
Please post the full output from the error. 

And likely, this is an issue with the repo (try running `qmk git-submodule` and see if that helps), or possibly an issue with modifications to the keymap.

agsb, 2024-12-01:
agsb@ID340:~/Gits/qmk_firmware$ bash
agsb@ID340:~/Gits/qmk_firmware$ qmk git-submodule
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
Removing 'lib/chibios-contrib/ext/mcux-sdk'
Synchronizing submodule url for 'lib/chibios'
Synchronizing submodule url for 'lib/chibios-contrib'
Synchronizing submodule url for 'lib/chibios-contrib/ext/mcux-sdk'
Synchronizing submodule url for 'lib/googletest'
Synchronizing submodule url for 'lib/lufa'
Synchronizing submodule url for 'lib/lvgl'
Synchronizing submodule url for 'lib/pico-sdk'
Synchronizing submodule url for 'lib/printf'
Synchronizing submodule url for 'lib/vusb'
Submodule path 'lib/chibios-contrib/ext/mcux-sdk': checked out '102c27d7cc2aaac87c133702e7829a105e0cb433'
agsb@ID340:~/Gits/qmk_firmware$ make ferris/sweep:default
QMK Firmware 0.26.11
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
Makefile:407: *** missing separator (did you mean TAB instead of 8 spaces?).  Stop.
agsb@ID340:~/Gits/qmk_firmware$

zaccaffey, 2024-12-01:
@agsb I am able to make without issue. Strange

agsb, 2024-12-01:
agsb@ID340:~/Gits/qmk_firmware$ git status
On branch master
Your branch is up to date with 'origin/master'.

It took 58.82 seconds to enumerate untracked files.
See 'git help status' for information on how to improve this.

nothing added to commit but untracked files present (use "git add" to track)
agsb@ID340:~/Gits/qmk_firmware$ git remote -v
origin  git@github.com:qmk/qmk_firmware.git (fetch)
origin  git@github.com:qmk/qmk_firmware.git (push)
agsb@ID340:~/Gits/qmk_firmware$

agsb, 2024-12-01:
Done git pull --rebase. Same problem. 
etc
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
  milc.set_metadata(version=__version__)
Makefile:407: *** missing separator (did you mean TAB instead of 8 spaces?).  Stop.

agsb, 2024-12-02:
Clean redo.
 git clone, qmk setup ,  git remote add upstream https://github.com/qmk/qmk_firmware.git  ,  git fetch origin  ,   git stash -u  ,  all ok.
but same result
make ferris/sweep:default ...
etc
/home/agsb/.local/lib/python3.12/site-packages/qmk_cli/script_qmk.py:18: UserWarning: milc.set_metadata has been deprecated, please use cli.milc_options() instead.
milc.set_metadata(version=version)
Makefile:407: *** missing separator (did you mean TAB instead of 8 spaces?). Stop.

Mistery

agsb, 2024-12-02:
qmk compile -kb ferris/sweep -km default  

works fine. Thanks for patience !