qmk/qmk_firmware #24614: qmk setup failed ][=--=[]]\

pallas42, 2024-11-18:


pallas42, 2024-11-20:
problem solved by downloading the [qmk toolchains](https://github.com/qmk/qmk_toolchains/releases) and adding to $PATH.

closed 2024-11-20 by pallas42

SerikDM, 2024-11-24:
Had same issue, solved with installing `avr-libc`

jirikrepl, 2025-01-11:
here are my two cents.

Before I saw this error I even did not have arm-none-eabi-gcc installed.
I had to manually install with brew.
then when running again `qmk setup` i saw this error

if you create a `test.c` file they it's possible to see why it is not compiling

```
int main() {
    return 0;
}
```

```
$ arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb --specs=nosys.specs --specs=nano.specs -o test.elf test.c
```

I uninstalled arm-none-eabi-gcc from brew and installed newest pkg on my macos from arm website

my error was that nosys.specs is not found.
I downloaded this software and installed it
https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

pkg on my mac was installed in 
`/Applications/ArmGNUToolchain/14.2.rel1/arm-none-eabi/bin/`

so in my `.zshrc` I added it on path `
export PATH="/Applications/ArmGNUToolchain/14.2.rel1/arm-none-eabi/bin/:$PATH"
`

and then test program can be compiled and `qmk setup` uses manually installed version and finish fine

fauxpark, 2025-01-12:
**Do not do this.** The ARM toolchain from osx-cross which the QMK setup process installs should work fine.

Typically the issue here is that `osx-cross/arm/arm-none-eabi-gcc@8` fails to install or link properly because something else is in the way. Homebrew core has a broken toolchain formula that users might be installing by mistake; Homebrew will refuse to link the proper toolchain as it collides with the existing one. Currently the QMK install script does not handle this, but on the develop branch I have changed it so that it nukes all of the known incorrect toolchains before attempting to install the right one (https://github.com/qmk/qmk_firmware/pull/24637).

There may also simply be something wrong with your Homebrew install that is causing the formula not to be symlinked - this is out of scope for QMK, and you'll need to go elsewhere for help fixing that. Running `brew doctor` is a good starting point.