qmk/qmk_firmware #24670: RGB matrix - hue_steps vs RGB_MATRIX_HUE_STEP  (multiple vs single)

fightforlife, 2024-12-03:
### Issue Description

I recently noticed that there is a difference in naming for the 4 STEP defines in json vs c.
- In json it is called e.g. "sat_steps: 8" (multiple), which suggests that there are 8 steps between no and full saturation. Increasing these steps would result in finer control.
- in C it is called e.g."RGB_MATRIX_SAT_STEP=8" (singular), which suggests that each time you increase/decreaste saturation, it will go 8 off 255 steps down or up. Increasing this step would result in less control.

The problem is, that the first interpretation and the second interpretation are contrary..
In the Code it looks like the second / C interpretation is correct.
https://github.com/qmk/qmk_firmware/blob/191172aa4e03081ec36cc918a6ac7ffe289162d6/quantum/rgb_matrix/rgb_matrix.c#L608

fightforlife, 2025-03-04:
Still valid

bintoro, 2025-03-04:
I noticed the same thing. And it’s not just the variable names; the info.json reference specifically documents the `*_steps` attributes as “The number of [...] adjustment steps”.

So, either the JSON attributes are incorrectly named and documented or there was supposed to be a transformation step like `RGB_MATRIX_VAL_STEP = RGB_MATRIX_MAXIMUM_BRIGHTNESS / rgb_matrix.val_steps` but which is currently missing.

clownfish-og, 2025-03-04:
this seems like a very valid issue and in fact I was mislead by this page in the docs as well

PR #25167 (2025-04-17): unify lighting step descriptions across docs
This attempts to resolve #24670 by clarifying the descriptions in the docs.

## Description

as titled.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

#24670 

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR  [... cut]
closed 2025-04-17 by waffle87