qmk/qmk_firmware #24675: Support more than 256 leds

mateuszdrwal, 2024-12-04:
I made a split keyboard with 4 leds under each key (10 for the 2 2U keys) and the results are [pretty spectacular](https://github.com/user-attachments/assets/8e028842-690f-4ba8-81c3-1d9bc073de4a). This sums up to 162 leds per half, or 324 leds on both. QMK currently uses a `uint_8` in most places to iterate over leds, so this required some patches which I want to polish up and contribute back.

I'm not intimately familiar with the codebase or nuances of QMK, so it would be great with some guidance on the best way to implement it for this project. I assume RAM is enough of a concern for the AVR platform that replacing all relevant `uint_8` with `uint_16` would be problematic? Would a `typedef uintX_t num_leds_t` then be best setting the type to uint16_t if `RGB_MATRIX_LED_COUNT > 255`, and using that type everywhere instead? Anything else I should know before putting time into implementing this properly?

drashna, 2024-12-06:
In general, yeah, AVR ram limitations are usually an issue.

I'm not sure of the best solution, but a typedef sounds like probably the best option here.  There is a similar thing for the layer state bitmask, for instance.

getreuer, 2024-12-06:
That board is gorgeous!

Maybe the LED index type can be defined conditionally based on `RGB_MATRIX_LED_COUNT`? It would be nice if it works automatically without needing an additional configuration. Perhaps:

```c
#if (RGB_MATRIX_LED_COUNT) <= 255
typedef uint8_t led_index_t;  // Continue using uint8_t by default.
#else
typedef uint_fast16_t led_index_t;  // If there are more than 255 LEDs.
#endif

bool SOME_EFFECT(effect_params_t* params) {
    LED_MATRIX_USE_LIMITS(led_min, led_max);
    for (led_index_t i = led_min; i < led_max; i++) {
       ...
```

mateuszdrwal, 2024-12-09:
Yep, that's what I'm thinking. `led_index_t` is probably a better type name than what I came up with.

christrotter, 2024-12-24:
I ran into this on my last board, but dang, your implementation of 'maxing out led count' is far more wondrous.

closed 2025-04-23 by github-actions[bot]

drashna, 2026-09-22:
I think I may have this working.   @mateuszdrwal @lazizbekravshanov would you be willing to give this commit a spin?
https://github.com/drashna/qmk_firmware/commit/4a2a73488c2c103936b5a13f8c555bc6e52ed1df