qmk/qmk_firmware #24682: [Bug] detected_host_os() broken

dlip, 2024-12-06:
### Describe the Bug

I use os detection to automatically switch between ctrl+v cmd+v depending on linux or mac using `detected_host_os()`
https://github.com/dlip/qmk_firmware/blob/dlip/keyboards/tenshi/keymaps/engram/keymap.c#L276

It broke in this revision: 305e7baa71ccecf8c126863a7f439361dd9cc750

### Keyboard Used

tenshi

### Link to product page (if applicable)

https://github.com/dlip/qmk_firmware/blob/dlip/keyboards/tenshi

### Operating System

MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/dane/code/qmk_firmware
Ψ Detected Linux (NixOS 25.05 (Warbler)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/dane/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: HEAD
Ψ Repo version: 0.26.7
Ψ - Latest HEAD: 2024-10-06 21:44:36 +1300 (305e7baa71) -- Allow for `get_hardware_id()` to be used for serial number. (#24053)
Ψ - Latest upstream/master: 2024-12-06 03:58:28 +0900 (09fdabf37c) -- Fix keymap.json error of keyboard/suikagiken/suika27melo (#24677)
Ψ - Latest upstream/develop: 2024-12-05 18:59:00 +0000 (55ea94832c) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2024-10-06 21:44:36 +1300 (305e7baa71) -- Allow for `get_hardware_id()` to be used for serial number. (#24053)
Ψ - Common ancestor with upstream/develop: 2024-10-06 21:44:36 +1300 (305e7baa71) -- Allow for `get_hardware_id()` to be used for serial number. (#24053)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.3.1
Ψ Found avr-gcc version 13.3.0
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

getreuer, 2024-12-06:
The OS Detection documentation describes [how to collect debugging data](https://docs.qmk.fm/features/os_detection#debug). I understand you are describing a regression where the detection used to work. Even so, maybe that data would help here.

drashna, 2024-12-10:
I haven't seen it break on my end.  However, I do check for macOS and iOS, as sometimes it does detect as iOS on my mac. 

Additionally, it might be worth printing out the detected os to console:
https://github.com/drashna/qmk_userspace/blob/master/users/drashna/drashna.c#L141-L212
This function does more than that, but you can see the prints.

dlip, 2024-12-11:
@drashna Before the merge it says "MacOS Detected" but after it says "iOS Detected". I think because your code swaps ctl/gui in either case, you didn't notice a difference.

tzarc, 2024-12-11:
Could also be an OS update realigning with the iOS implementation.
Pretty sure @drashna is on an older-spec Hackintosh.

drashna, 2024-12-11:
I have an Intel based hackentosh running the latest version of Sequoia.  

And it does report macOS. But occasionally, yes, it does report iOS. It's why I have it set for both.

drashna, 2024-12-11:
Aah, I see.  Currently, it's not properly filtering macOS, because Apple changed some stuff a while back (that or the updated USB endpoint code is more efficient and picks up more data, skewing the results.  It does show up as iOS on master/develop, for me, but I have some custom bodges for detection. 

However, this is part of what it means by "best guess".

dlip, 2024-12-12:
@drashna Yes, that does fix it for me!

drashna, 2024-12-12:
> @drashna Yes, that does fix it for me!

glad to hear it!  Though, could you collect the debug information, as mentioned in this comment: https://github.com/qmk/qmk_firmware/issues/24682#issuecomment-2524231579

dlip, 2024-12-13:
@getreuer Here is the output without @drashna 's fixes:

```
HID console connected: Dane Lipscombe tenshi (BAAE:0212:0100)
> iOS Detected
> i: 0, wLength: 0x02
> i: 1, wLength: 0x0E
> i: 2, wLength: 0x02
> i: 3, wLength: 0x1E
> i: 4, wLength: 0x02
> i: 5, wLength: 0x42
> i: 6, wLength: 0xFF
```

drashna, 2024-12-14:
For the Mac, was this on apple silicon?  Or an intel mac?  (and sequoia/15.x)?

getreuer, 2024-12-16:
Thank you for sharing that, @dlip!

I'm no expert on USB, but noticed this bit of news that seems potentially relevant: [Apple released update 14.4.1 in March 2024](https://www.macrumors.com/2024/03/25/apple-releases-macos-sonoma-14-4-1/) to fix several problems with USB hubs. I wonder if that modified the fingerprint.

> According to Apple's release notes, the macOS Sonoma 14.4.1 update fixes an issue that could cause USB hubs connected to external displays not to be recognized. It also addresses an issue that could cause apps with Java to quit unexpectedly, and it fixes an issue that could cause Audio Unit plug-ins for professional music apps not to open.

dlip, 2024-12-17:
@getreuer Not sure if its useful info, but I did have it plugged into a usb hub. I tried plugging it in directly but it still detected as ios

PR #24708 (2025-01-29): [OS Detection] Improve MacOS detection

## Description

Fix detection for macOS by making the detection more flexible.

## Types of Changes

- [x] Core
- [x] Bugfix

## Issues Fixed or Closed by This PR

* Closes #24682

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have made the appropriate changes.
- [ ] My change requires a change to the documentation.
- [x] I have up [... cut]
closed 2025-01-29 by tzarc