qmk/qmk_firmware #24685: JSON schema is broken.

uk-taniyama, 2024-12-07:
"," is missing.
https://github.com/qmk/qmk_firmware/blob/a12ff287eed88d928430b63f4f7faff3b6fb5ebc/data/schemas/keyboard.jsonschema#L355-L358

There is an extra ",".
https://github.com/qmk/qmk_firmware/blob/a12ff287eed88d928430b63f4f7faff3b6fb5ebc/data/schemas/user_repo_v0.jsonschema#L10-L13
https://github.com/qmk/qmk_firmware/blob/a12ff287eed88d928430b63f4f7faff3b6fb5ebc/data/schemas/user_repo_v1.jsonschema#L11-L13
https://github.com/qmk/qmk_firmware/blob/a12ff287eed88d928430b63f4f7faff3b6fb5ebc/data/schemas/user_repo_v1_1.jsonschema#L12-L14

fauxpark, 2024-12-07:
These are currently read as HJSON, so it is technically not an issue at the moment. There is an open PR which fixes the syntax issues as part of its changeset: #24594

Buckwich, 2024-12-08:
If you want to use json-schemas that follow the json and json-schema specification I have extracted them in a seperate Repo until my PR https://github.com/qmk/qmk_firmware/pull/24594 is merged: https://github.com/kcf-hub/qmk-types 

With those schemas, ref resolution is also adjusted so that tools like vs-code can directly resolve them by relative paths