qmk/qmk_firmware #24691: [Bug] SEND_STRING() adds spurious characters at the end

girishji, 2024-12-08:
### Describe the Bug

I do `SEND_STRING("for(int i = 0; i < ; i++) {}"` and I getr `SEND_STRING("for(int i = 0; i < ; i++) {}6`". This occurs for strings over a certain length. It always adds some number followed by ```.

### Keyboard Used

custom

### Operating System

rp2040 keyboard and macos

### qmk doctor Output

```text
❯ qmk doctor    
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /Users/gp/git/qmk_firmware
Ψ Detected macOS 13.4.1 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.27.0
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
☒ Can't find avr-gcc in your path.
Would you like to install dependencies? [Y/n] n
Ψ Found arm-none-eabi-gcc version 12.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
⚠ Unknown version for avr-gcc
Ψ Found avrdude version 7.2
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b33)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ Major problems detected, please fix these problems before proceeding.
Ψ Check out the FAQ (https://docs.qmk.fm/#/faq_build) or join the QMK Discord (https://discord.gg/qmk) for help.

```
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2024-12-11:
Do you have any software installed on the system that may intercept keypresses, such as karabiner? 
Also, what keyboard layout are you using?

girishji, 2024-12-12:
No I do not have karabiner installed. It is a clean macos system. The spurious keys get inserted after the usb report for all the typed keys is sent is sent. I have not debugged this further. Here is my keyboard:

https://github.com/girishji/keyboard_qmk/tree/main/opticalkb/rev3

girishji, 2025-01-05:
Found the problem. This is not a qmk issue. The long latency in sending string through send_string() was causing saturation in a few phototransistors in my optical keyboard. Easily fixed.

closed 2025-01-05 by girishji