qmk/qmk_firmware #24695: [Feature Request] Bluetooth: Don't lose the very first key presses while the keyboard is in deep sleep

bam80, 2024-12-08:
### Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

Hi, let me quote [the request for ZMK](https://github.com/zmkfirmware/zmk/issues/2686) here, it apparently applies to QMK also:

> Hi,
I'm new to ZMK, but a long time BT keyboards user.

> One behavior driving me nuts is the loosing of the first key-presses if the keyboard is in deep sleep (and thus disconnected itself from BT host).
> As it's turned out, one half of the problem is a regression in Linux BT stack (Bluez), when the key events gotten by host are just discarded until the keyboard fully re-connects (from host POV). The optional fix (workaround?) for Bluez was recently introduced:
> https://github.com/bluez/bluez/issues/737
> 
> That still wasn't the end of the story, unfortunately:
> apparently, some BT keyboards (even modern ones, with BT 5.0) do not pass the key events to host at all, if they were in deep sleep!:
> https://github.com/bluez/bluez/issues/737#issuecomment-2509751005
> That was quite unexpected to me, because even ancient BT 3.0 "Microsoft" keyboard didn't have the issue, as well as the other cheap one.
> 
> I really upset about this situation. The proper behavior in this case is essential to me.
> My only hope is ZMK (and others free keyboard firmwares) do that properly.
> 
> Any insight how things are really going?
> Could we implement the feature in ZMK, if it doesn't do that yet?
> Thanks.

tzarc, 2024-12-09:
The closest anyone has been to this area recently has been @KarlK90, so I’ll defer to their familiarity in this area (if they have any comments at all).

Ultimately this is low priority for the QMK team who have a whole host of other things going on as well — your best bet is to try to rectify it yourself.

bam80, 2024-12-13:
Thanks for reply @tzarc, I refactored the feature request to a bug, as apparently there is really no reason for a keyboard to hold these keypresses.

drashna, 2025-01-25:
The problem isn't so much as the keyboard holding the keypresses but rather the implementation of bluetooth.  The report is sent and then discarded.  You'd basically need to buffer keypresses until bluetooth connection is confirmed re-established and then send.  And you'd probably need a bunch of additional code to intelligently handle this buffer (eg, discarding after a certain time, etc).  

While it's possible to accomplish this, most likely... the wireless stack isn't really complete, as is, and implementation for this may be per "provider"/driver (though, probably possible as a standardized option, but not sure)

bam80, 2025-04-26:
Please don't close

bam80, 2025-07-26:
Please reopen.

tzarc, 2025-07-27:
As stated before, this is low priority and you should tackle it yourself, especially given that it's Bluetooth.
If it goes stale again I'm just going to close it.

tzarc, 2025-07-27:
We can't cater for everyone, there are other high priority things to look at. Leaving things open which will never get looked at by the team wastes everyone's time.

Will leave it open until the next stalebot iteration over this issue, up to you to figure out how to get it addressed in that time.

bam80, 2025-07-27:
That would be impolite , but if you decided to do that, you can close now, as nothing will change from my side.