qmk/qmk_firmware #24700: qmk setup will not accept my Github Username because it has a hyphen in it.

Orphan-Crippler, 2024-12-10:
### Issue Description

Just like the title says, when I am trying to make a new keymap qmk is will not let me use my github username because there is a hyphen in it.

waffle87, 2024-12-10:
This is not an "issue", the error you're present with:
```shell
☒ Keymap names must contain only a-z, 0-9 and _! Please choose a different name.
```
Is simply a requirement of the build system. You must choose a keymap name that does not have a hyphen in it.

The keymap name does not need to match your Github username nor does it have any affect what you choose to name it, so long as it follows the expected regex.

Orphan-Crippler, 2024-12-10:
Thank you, I thought it was asking for my GitHub username. It is a little misleading as the last thing it displays is asking about that. Maybe this could be changed to something better?

❯ qmk new-keymap -kb crkbd
Ψ Generating a new keymap

Name Your Keymap
Used for maintainer, copyright, etc

Your GitHub Username?  [Orphan-Crippler]

waffle87, 2024-12-12:
Hopefully a bit clearer in #24701.

closed 2024-12-12 by waffle87

Orphan-Crippler, 2024-12-12:
Yea, that's definitely better!

Thank You!