qmk/qmk_firmware #24741: [Bug] VICE fails to register Mod-Tap keystrokes

jacereda, 2024-12-22:
### Describe the Bug

Missed keystrokes when running https://vice-emu.sourceforge.io/

### Keyboard Used

crkbd

### Link to product page (if applicable)

https://github.com/pashutk/chocofi

### Operating System

Windows and NixOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/jacereda/src/qmk_firmware
Ψ Detected Linux (NixOS 25.05 (Warbler)).
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /home/jacereda/src/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.27.1
Ψ - Latest master: 2024-12-22 00:14:59 -0500 (f3fe59afe2) -- fix: mechwild/bb65 coordinate mix-up (#24738)
Ψ - Latest upstream/master: 2024-12-22 00:14:59 -0500 (f3fe59afe2) -- fix: mechwild/bb65 coordinate mix-up (#24738)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-12-22 00:14:59 -0500 (f3fe59afe2) -- fix: mechwild/bb65 coordinate mix-up (#24738)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.3.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 13.3.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Additional Context

Looks like the default value for `TAP_CODE_DELAY` is too low, I defined it to 40 ms and it fixes the issue. Also tried with 20 ms but it was still missing keystrokes from time to time. Would it be a good idea to increase the default value?

tzarc, 2024-12-22:
This sounds more like an issue with the software you’re using rather than with QMK. Do these problems manifest on the same machine and setup, using key tester software instead of this VICE app?

jacereda, 2024-12-23:
That could well be the case, but the same problem is present in https://retroshowcase.gr/cpcbox-master/ or https://virtualconsoles.com/online-emulators/zx-spectrum/ or https://virtualconsoles.com/online-emulators/c64/ or https://archive.org/details/c64_basic

You don't get press/release events with 0 ms difference between them in normal keyboards. You might be right from a theoretical viewpoint, but I guess it's easier to fix the problem here than to try to fix the world.

tzarc, 2024-12-23:
If this were a far more widespread issue there would be many more reports of this stuff being problematic. Using tools such as `evtest` on Linux should show that these reports are received by the OS — after that it’s up to the operating system and other application layers to decide if they should act.

Your suggested fix doesn’t address the root cause but rather works around it — the application layers expected to propagate the event seem to be failing. If the keyboards weren’t reporting at all then it’d be a QMK issue. 

I’d say this is likely a change to be made to your keymap, rather than promoting this as a default change.

tzarc, 2024-12-23:
For context, a tap code delay of 40 in my keymap would result in single Unicode characters taking up to 320ms to register. Without your fix I can type as if they’re normal characters at normal typing speed even though they result in upwards of 8 different reports being sent. With your fix I’d get 3 keypresses per second — anything higher than that would mean physical keypress events being missed.

In my case I’d advocate for no delay. Who takes precedence?

This is a good reason why this stuff is configurable at the keymap level — different users with different use cases can tweak as they see fit.