qmk/qmk_firmware #24746: [Bug]

am-zed, 2024-12-24:
### Describe the Bug

qmk breaks with Python 3.13 but works fine with Python 3.12.

qmk commands fail with claiming "`AttributeError: 'MILC' object has no attribute 'milc_options'`" in "`/usr/lib/python3.13/site-packages/qmk_cli/script_qmk.py`" line 18 "`milc.cli.milc_options(version=__version__)`".

Commenting out the line made it work with Python 3.13.

### Keyboard Used

crkbd/rev4_1

### Operating System

Arch Linux x86_64 - 6.13.0-rc3-1-mainline

### qmk doctor Output

```text
# qmk dcotor gives:
Traceback (most recent call last):
  File "/usr/bin/qmk", line 5, in <module>
    from qmk_cli.script_qmk import main
  File "/usr/lib/python3.13/site-packages/qmk_cli/script_qmk.py", line 18, in <module>
    milc.cli.milc_options(version=__version__)
    ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'MILC' object has no attribute 'milc_options'

# but commenting out line 18 from /usr/lib/python3.13/site-packages/qmk_cli/script_qmk.py:
Ψ QMK Doctor is checking your environment.
Ψ CLI version: unknown
Ψ QMK home: /home/azed/qmk_firmware
Ψ Detected Linux (Arch Linux).
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /home/azed/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.27.1
Ψ - Latest master: 2024-12-13 20:42:47 +1100 (597bbcb567) -- Add licensing violations page. (#24697)
Ψ - Latest upstream/master: 2024-12-24 02:22:22 -0500 (276f84a9a8) -- add chlx piche (#24252)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2024-12-13 20:42:47 +1100 (597bbcb567) -- Add licensing violations page. (#24697)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.2.0
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.1.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

Vial

### Additional Context

Most recent change was updating Python from 3.12 to 3.13:

[10-11T18:03:56+0200] [ALPM] upgraded python (3.12.6-1 -> 3.12.7-1)
[2024-12-24T00:20:31+0100] [ALPM] upgraded python (3.12.7-1 -> 3.13.1-1)

Tried many versions of Python 3.12.* using virtualenv, but still same issue with qmk commands.

tzarc, 2024-12-24:
QMK works fine with 3.13. Update your copy of `milc`.

am-zed, 2024-12-24:
@tzarc, that's the case:

```sh
$ python --version
Python 3.13.1

# installed python paths
$ yay -Ql python
...
python-milc /usr/lib/python3.13/site-packages/milc-1.8.0-py3.13.egg-info/
...

# python-milc installed version
$ yay -Ss python-milc
extra/python-milc 1.8.0-3 (51.6 KiB 238.2 KiB) (Installed)
    Opinionated Batteries-Included Python 3 CLI Framework.

# python-milc package details
$ yay -Qi python-milc
Name            : python-milc
Version         : 1.8.0-3
Description     : Opinionated Batteries-Included Python 3 CLI Framework.
Architecture    : any
URL             : https://milc.clueboard.co/#/
Licenses        : MIT
Groups          : None
Provides        : None
Depends On      : python-appdirs  python-argcomplete  python-colorama  python-halo
Optional Deps   : None
Required By     : qmk
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 238.18 KiB
Packager        : Jelle van der Waa <jelle@archlinux.org>
Build Date      : Thu 21 Nov 2024 02:51:58 PM CET
Install Date    : Tue 24 Dec 2024 08:22:22 PM CET
Install Reason  : Installed as a dependency for another package
Install Script  : No
Validated By    : Signature

# python-milc paths
$ yay -Ql python-milc
...
python-milc /usr/lib/python3.13/site-packages/milc-1.8.0-py3.13.egg-info/
...
python-milc /usr/lib/python3.13/site-packages/milc/
...
```

am-zed, 2024-12-24:
Installing using pip fixed the issue; now I have milc version 1.9.0 while system package offers version 1.8.0.
Tried `import milc` and `milc.cli.<TAB>` in **bpython** and the attributes were different between the two versions.

tzarc, 2024-12-24:
Yep, Arch is out of date and we’re not the maintainers of those packages.

closed 2024-12-24 by am-zed