qmk/qmk_firmware #24760: [Feature Request] Iterative quadrant-based mouse pointing

windo, 2024-12-29:
### Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I've never enjoyed mousekeys, even with accelerations (although now that I think of it, I don't think I've really given it a very serious attempt). As a vim user, it's irritating to have to hold down keys for various amounts of time to point at things. But sometimes (even with all the vim browser extensions etc) you still need to do it.

Anyway, I came up with an idea (or maybe remembered it, if it's been tried before) to come up with an addressing scheme for the entire screen, so that you could learn muscle memory for pointing at any position on screen with some small number of keystokes (I went for 3-4 ish).

With the digitizer support in QMK, I can move the pointer to absolute positions, and I went with the basic premise of dividing the screen into quadrants (actually one per 8 directions, so octrants?) and have the pointer jump from the center of the screen to one of those quadrants with one key press. But at the same time, reduce the scale of the movement to be restricted within the quadrant, so that the next keypress would further move towards the intended destination.

So for example, pressing w (for up) just once would get you to 2/3 way from the center of the screen towards the upper edge, and pressing 3 times would take you all the way. But you could also press w-s (up-down) to land a bit past 1/3 of the screen.

I have a visual aid for how to get to various points on the screen with wasd (directions ) + qezc (diagonals):

https://p6drad-teel.net/~siim/navigation/motions.png

The cursor update rules I went with are these:

https://github.com/windo/qmk_userspace/blob/main/keyboards/ergodox_ez/keymaps/windo/quadrant.c#L25

And I have a wonky in-browser implementation where I was trying this out first, that you can poke at if this sounds interesting at all:

https://p6drad-teel.net/~siim/navigation/

(wasd+qezc for moving the cursor, 012 for how many keystrokes forward to show where the cursor would go, b to toggle the screenshot background on/off, r to reset the cursor back to start of pointing).

I've done a bit of pointing with it for a few days and I don't especially despise it so far (eg: I manged to use gimp with keyboard only to crop that screenshot of motions). Does something like this sound like it might be a useful alternative for mousekeys?

Kriechi, 2024-12-29:
Reminds me about this macOS application: `mouseless`:
https://mouseless.click/
https://youtu.be/J0rwQVNQkHM

windo, 2024-12-29:
Oh nice, he looks super fast in using that grid. That visual overlay seems very helpful for learning, but I reckon after some deliberate practice you don't really need to see the grid any more.

raymon-roos, 2024-12-30:
I have used [warpd](https://github.com/rvaiya/warpd) for this in the past, which does something similar. It would be cool to accomplish this through QMK.

drashna, 2025-01-10:
mousekeys and pointing device is completely ignorant of the host system, in its entirety. 

However, this is much more likely doable via the digitizer feature.

windo, 2025-01-13:
Inspired by warpd I added an implementation (quite possibly in a hacky way) to also support for an alphabet-sized grid input. My current feeling is that both of these methods are very nice for quickly getting your cursor to the appriximately right part of the screen.

But then with alphabet grid, you'd always need to follow up with a more precise method to point at any smaller UI elements. Using mouse keys tuned for local cursor adjustments would be an obvious choice, or following up with the 9x9 directional grid would also work.

The 9x9 grid can be used to point at pretty small things, but if you've misjudged somewhere on the path then it's not superr forigiving. I've currently added an `undo` and a `set scale for local movement` keys but neither of those I've really found especially helpful so far. So maybe those should be scrapped. Obviously, mouse keys could work to deal with this as well.

tzarc, 2025-04-14:
Closing as #24811 wasn't continued by the OP.

closed 2025-04-14 by tzarc