qmk/qmk_firmware #24763: How to create compile_commands.json database on every build.

Antecer, 2024-12-30:
### Issue Description

I just know run `qmk generate-compilation-database -kb <keyboard> -km <keymap>` can build database.
but it will clean all build files.
how to create compile_commands.json database on every build.
when i run `make <keyboard>:<keymap>` it will create `compile_commands.json` and build firmware.

tzarc, 2024-12-30:
`qmk compile —compiledb -kb <keyboard> -km <keymap>`

Antecer, 2024-12-30:
> `qmk compile --compiledb -kb <keyboard> -km <keymap>`

here have multiple keyboard layouts.
i will frequently compile multiple firmwares with different configurations for debugging purposes.
So, how can I add this command to my build process?

Antecer, 2024-12-30:
> `qmk compile --compiledb -kb <keyboard> -km <keymap>`

Yes, it works fine.
But the input is a bit troublesome. 
How can I add it into `make <keyboard>:<keymap>`, it will easy to input.

tzarc, 2024-12-30:
Compilation database needs to clean `.build/`. If you want incremental, then build the compilation DB first then don’t re-run it with the command argument.

You can’t get that with `make`, at all.

Antecer, 2024-12-30:
Ok, I figured out how to do this.

closed 2024-12-30 by Antecer