qmk/qmk_firmware #24810: [Bug] Multiple definitions of pmw33xx_pointing_device_driver'

Schievel1, 2025-01-12:
### Describe the Bug

When compiling the tractyl manuform with the default keymap I get a whole bunch of 
``` | /usr/libexec/gcc/arm-none-eabi/ld: .build/obj_handwired_tractyl_manuform_5x6_right_f411_default/keyboards/handwired/tractyl_manuform/5x6_right/5x6_right.o:(.rodata+0x0): multiple definition of `pmw33xx_pointing_device_driver'; .build/obj_handwired_tractyl_manuform_5x6_right_f411_default/keyboards/handwired/tractyl_manuform/tractyl_manuform.o:(.rodata+0x0): first defined here
```
also for 
```
 | /usr/libexec/gcc/arm-none-eabi/ld: .build/obj_handwired_tractyl_manuform_5x6_right_f411_default/protocol/chibios/usb_util.o:(.bss+0x0): multiple definition of `usb_endpoint_interface_lut'; .build/obj_handwired_tractyl_manuform_5x6_right_f411_default/protocol/chibios/usb_main.o:(.bss+0x0): first defined here
```

during linking. 

I think this is due to the new gcc 14 that I am using, though I haven't tested it with an earlier version of gcc yet. 
When compiling with  `CFLAGS+="-lc -specs=nosys.specs -Wl,--allow-multiple-definition"` it works. 
Where I put in `-lc -specs=nosys.specs` because of https://github.com/qmk/qmk_firmware/issues/19071

### Keyboard Used

Tractyl Manuform

### Operating System

Gentoo Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.2
Ψ QMK home: /home/pascal/dev/qmk_firmware_dm_r_track
Ψ Detected Linux (Gentoo Linux).
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /home/pascal/dev/qmk_firmware_dm_r_track/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.27.8
Ψ - Latest master: 2025-01-10 20:59:33 +0000 (35c6520560) -- Bump nanoid from 3.3.7 to 3.3.8 in /builddefs/docsgen (#24713)
Ψ - Latest upstream/master: 2025-01-10 20:59:33 +0000 (35c6520560) -- Bump nanoid from 3.3.7 to 3.3.8 in /builddefs/docsgen (#24713)
Ψ - Latest upstream/develop: 2025-01-11 01:29:58 +0000 (62e98327d2) -- Azoteq - improve I2C behaviour while polling. (#24611)
Ψ - Common ancestor with upstream/master: 2025-01-10 20:59:33 +0000 (35c6520560) -- Bump nanoid from 3.3.7 to 3.3.8 in /builddefs/docsgen (#24713)
Ψ - Common ancestor with upstream/develop: 2025-01-10 20:59:33 +0000 (35c6520560) -- Bump nanoid from 3.3.7 to 3.3.8 in /builddefs/docsgen (#24713)
Ψ CLI installed in virtualenv.
☒ Can't find dfu-programmer in your path.
Would you like to install dependencies? [Y/n] n
Ψ Found arm-none-eabi-gcc version 14
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 8.5.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.2
⚠ Unknown version for dfu-programmer
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ Major problems detected, please fix these problems before proceeding.
Ψ Check out the FAQ (https://docs.qmk.fm/#/faq_build) or join the QMK Discord (https://discord.gg/qmk) for help.

Note the problems are only the missing dfu-programmer. Which I don't need to compile.
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2025-01-12:
Don't use CFLAGS in the command line.  It breaks things, as even with "+=", it actually just replaces it, completely. Eg, it wipes out any values that have been assigned to it previously. 

Add it to a `rules.mk` file, and then try compiling.

Schievel1, 2025-01-13:
Strangely enough I got "arm-none-eabi-gcc (Gentoo 14.2.1_p20241221 p7) 14.2.1 20241221" on another machine and it works. Still needs `EXTRAFLAGS+="-U_FORTIFY_SOURCE"` because of Gentoos GCC being hardened a little. I am trying to figure out what is different with the GCCs on those systems. 

I will give it a try with CFLAGS in rules.mk on the machine where it's not working later.

sigprof, 2025-01-13:
Actually this looks like a bug in https://github.com/qmk/qmk_firmware/blob/26f898c8a538b808cf506f558a9454f7f50e3ba6/drivers/sensors/pmw33xx_common.h#L108
The `extern` keyword is missing from the declaration, therefore it is actually recognized as a “tentative definition” in every compilation unit that includes `pmw33xx_common.h`.  Normally `CFLAGS += -fcommon` added in `builddefs/common_rules.mk` should hide that problem; maybe you are losing that flag due to some broken override of `CFLAGS`, or the handling of `-fcommon` for `const` data changed in GCC 14.

Schievel1, 2025-01-13:
I think the first thing is the case. 
`qmk compile -kb handwired/tractyl_manuform/5x6_right/f411 -km pascal -e CFLAGS+="-lc -specs=nosys.specs"` does not work, but `qmk compile -kb handwired/tractyl_manuform/5x6_right/f411 -km pascal -e EXTRAFLAGS+="-U_FORTIFY_SOURCE"` does. 
Also `qmk compile -kb handwired/tractyl_manuform/5x6_right/f411 -km pascal -e CFLAGS+="-lc -specs=nosys.specs -fcommon"` works.

closed 2025-01-16 by Schievel1