qmk/qmk_firmware #24853: Update for 'A-JAZZ AKC084' (A-JAZZ AKP846)

Programmer86, 2025-01-22:
### Issue Description

Hello everyone!

I recently purchased an **AJAZZ AKP846** keyboard and attempted to flash it with the standard QMK firmware. In the web configurator, I selected the model _'A-JAZZ AKC084'_, customized a few keys, and flashed the device. However, I encountered several issues:

**Keyboard Indicators:**
The standard firmware includes indicators for CapsLock and ScrollLock. After flashing, these indicators changed to CapsLock and NumLock, even though there is no NumLock key on the keyboard.

**Layout Mismatch:**
The standard layout does not match the keys _Pause_, _Del_, _End_. This discrepancy is visible in the image in the Readme and on the product page in the store.

**BootLoader Entry Instructions:**
To enter BootLoader mode, you need to press _Esc_ and reconnect the keyboard, instead of Enter as stated in the instructions.

Can I make changes and update the firmware for the 'A-JAZZ AKC084' model to help new users avoid these issues? Or are '**A-JAZZ AKC084**' and '**A-JAZZ AKP846**' different keyboard models?

Thank you!

tzarc, 2025-01-22:
Not sure where you're getting your firmware from but it's not a QMK-supported board.

fauxpark, 2025-01-22:
`ajazz/akc084` does exist in the repo, although it is very confusing as the readme links to the product page for the AKP846 (which looks identical to both the layout and the readme image), and I get no hits for the AKC084. So they might be the same board.

Programmer86, 2025-01-22:
Sorry if I don't understand something, but here it is:
https://browse.qmk.fm/#/keyboard/a_jazz/akc084
https://github.com/qmk/qmk_firmware/tree/master/keyboards/a_jazz/akc084

fauxpark, 2025-01-22:
For the indicators, yes, that should be changed, although I'm not sure where the Scroll Lock LED would be as there is also no Scroll Lock key either.

The default keymap and bootloader instructions in the readme do look like they need adjustment as well.

PR #24868 (2025-02-16): Update for 'A-JAZZ AKC084' ('A-JAZZ AKP846')

Fix indicators, image, bootloader instructions and default keymap for 'A-JAZZ AKC084'

## Description

**Keyboard Indicators:**
By default, the new keyboard has indicators for CapsLock and ScrollLock. After flashing, these indicators changed to CapsLock and NumLock, even though there is no NumLock key on the keyboard.

**Layout Mismatch:**
The standard layout does not match the keys Pause, Del, End. This discrepancy is visible in the image in the Readme and on the product page in the store.

**BootLoader Entry Instructions:**
To enter BootLoader mo [... cut]
closed 2025-02-16 by zvecr