qmk/qmk_firmware #24862: [Feature Request] Oneshot mods with mouse keys

gruvw, 2025-01-24:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

### Description

Hello, first let me thank all the contributors to this awesome firmware. I'm using it to create a custom keyboard used for accessibility purposes.

I was so thrilled to be finally be able to control my mouse again with simple keys (mouse layer). 
It is working very smoothly.

However, as I am using oneshot mods on a designated layer (I have a very limited amount of keys), I noticed one thing: I was unable to **use `ctrl+click` in my browser to open a link in a new tab**.
To be clear and complete, I am using `add_oneshot_mods` with ctrl modifier along with a click on the `MS_BTN1`.
I did not try it with a conventional held modifier, but I suspect the same problem might be present.

I marked this as a Feature Request because I did not find a way to have that ctrl + mouse click behavior.
I am sorry if the feature exists, and I did not find it, or if this issue is a duplicate (did not find a similar one by myself).
Of course ctrl is just an example, I am generally talking about **oneshot modifiers combined with mouse keys** (as shift+click would also be useful when I need to highlight text).

Hope that you can help, thank you so much for your help.
I am willing to but a 50$ donation bounty to implement this feature (in case it is missing) into QMK if that is allowed.

drashna, 2025-02-27:
The relevant code is in `action_util.c`, but basically .... mousekeys don't call `send_keyboard_report` which is what checks, applies and clears the oneshot mod status.  

A simple option may be to edit action.c, and the "register_mouse" to also run `send_keyboard_report`.  I'm not sure that will work, though, TBH.

gruvw, 2025-05-29:
Could we mark this feature as `to do`? I think it's currently missing on the firmware, and it isn't documented as such.

I think QMK should support this functionality.

tzarc, 2025-05-29:
Commenting will have un-stale’d the issue next time it runs. Adding “to do” doesn’t signify maintainer intent as it’s quite honestly not on our radar priority-wise. It’s not on any of our “to do” lists.

The onus is on you or someone else in the community to implement this feature — if it’s here as a perpetual placeholder with nobody willing to implement it, it’s of no use to anybody.