qmk/qmk_firmware #24926: Jurisdiction for considering songs in/out of copyright?

Laogeodritt, 2025-02-17:
### Issue Description

Hello!

I'm aware that various copyrighted songs were purged from this repository at one point (see quantum/audio/song_list.h). The only reference I could find to this was issue #5660, and I didn't find any specific policy on docs.qmk.fm or in the issues. I figured posting here would be best for future discoverability, vs. Discord or the OLKB subreddit.

I'm working on a special-use macropad layout for the Planck in which, as a specific reference, I want to use a melody from Sibelius's Violin Sonata in D minor, op. 47. The piece is public domain in the US and Canada, but not in the EU for example. (Published in 1905, Sibelius died in 1957.) I'd obviously like to include it in my pull request as a feature of the keymap.

My question is: Under what jurisdiction is the QMK project, for the purpose of determining copyright status of songs? OLKB appears to be US-based, as is GitHub as the repo's host, so is public domain in the US sufficient?

Thanks!

Laogeodritt

tzarc, 2025-02-17:
I'm not sure we really have an answer to that -- QMK Firmware is trademarked in the USA but that's about the extent of things, officially.

That said, we're no longer accepting user keymaps, so you're best off keeping a copy in your fork of QMK Firmware, or an associated external QMK Userspace repository.

Laogeodritt, 2025-02-17:
Thanks for the response!

re: user keymaps, ah, understood! I remembered this repository accepting user keymaps when I started using the Planck some ≈2 years ago and assumed that my recollection was correct/that this hadn't changed. I'll plan to share it via my own fork, then.

tzarc, 2025-02-17:
You may wish to give [external userspace](https://docs.qmk.fm/newbs_external_userspace) a go, it’ll allow you to keep your keymap outside of the qmk_firmware repo, simplifying upgrades and the like.

Laogeodritt, 2025-02-19:
Thank you! Aye, I found that documentation after you mentioned userspace repositories in your first response; currently moving my keymaps over to one. It's certainly always annoying to maintain non-upstreamed changes in a fork and keeping the fork synced to upstream, so this seems extremely useful.

closed 2025-02-26 by Laogeodritt