qmk/qmk_firmware #24935: Unclear whether the Lemokey (a gaming brand of Keychron) X0 QMK Wired Mechanical Gaming Keypad is supported

xuhdev, 2025-02-20:
### Issue Description

Based on the repository and the [supported keyboard list](https://browse.qmk.fm/#/), it looks like Keychron generally have their keyboard directly supported upstream here.

But I couldn't find out the support status of [Lemokey X0 QMK Wired Mechanical Gaming Keypad](https://www.keychron.com/products/lemokey-x0-qmk-wired-mechanical-gaming-keypad) (Amazon still has some in stock). I searched for "Lemokey" but it doesn't appear anywhere in code. Could this be an oversight?

(Is there a generally guideline on how to find the support status before purchasing? The supported keyboard list only contains identifiers, which can be very different from human readable names.)

tzarc, 2025-02-20:
You want https://github.com/Keychron/qmk_firmware/ for that brand. Future queries should be directed there.

yeroca, 2025-02-21:
None of the lemokey products have upstream configurations.

I have an unofficial, forward-ported version of the X0 config I can provide if you are interested.  I've been using it for a month or so without issues.