qmk/qmk_firmware #24953: [Bug] mod keys do not disable auto mouse layer

caut110, 2025-02-25:
### Describe the Bug

### How to reproduce:

1. set auto mouse layer
    https://docs.qmk.fm/features/pointing_device#pointing-device-auto-mouse
2. trigger auto mouse layer by using pointing device
3. mouse layer stays active after pressing a mod key (eg, `KC_RGUI`)

expected: disable auto mouse layer when a mod key is pressed.

### Use case:

1. set `q` key to be `MS_BTN4` on mouse layer
2. use pointing device to trigger auto mouse layer
3. press `Super + q` -> `Super + mouse btn4` is send

---

The cause seems to be [1].

In case it is a feature and not a bug,
is there a way to disable auto mouse layer when a mod key is pressed in `keymap.c`? 

[1] https://github.com/qmk/qmk_firmware/blob/6d0e5728aa61b442885d48caf49d29e5c60e8197/quantum/pointing_device/pointing_device_auto_mouse.c#L338

### Keyboard Used

halcyon kyria

### Link to product page (if applicable)

https://splitkb.com/products/halcyon-kyria

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.5
Ψ QMK home: /home/foo/qmk_firmware
Ψ Detected Linux (NixOS 25.05 (Warbler)).
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /home/foo/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Testing userspace candidate: /home/foo/qmk_userspace -- Valid `qmk.json`
Ψ QMK userspace: /home/foo/qmk_userspace
Ψ Userspace enabled: True
Ψ Git branch: master
Ψ Repo version: 0.27.13
Ψ - Latest master: 2025-02-20 14:07:18 +0000 (6d0e5728aa) -- Bump JamesIves/github-pages-deploy-action from 4.7.2 to 4.7.3 (#24933)
Ψ - Latest upstream/master: 2025-02-20 14:07:18 +0000 (6d0e5728aa) -- Bump JamesIves/github-pages-deploy-action from 4.7.2 to 4.7.3 (#24933)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2025-02-20 14:07:18 +0000 (6d0e5728aa) -- Bump JamesIves/github-pages-deploy-action from 4.7.2 to 4.7.3 (#24933)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 12.3.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.2.1
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-02-17 19:20:06 +0000 --  (be44b3305f)
Ψ - lib/chibios-contrib: 2024-04-03 20:39:24 +0800 --  (77cb0a4f)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2025-02-25:
Technically, not a bug.  You quoted the line right after `// Skip Mod keys to avoid layer reset`, which indicates that this is by design. 
Unfortunately, no, this isn't configurable currently, so you'd either have to use a custom macro for mods to force it, or edit the lines in question.

caut110, 2025-02-25:
Thanks for the pointers.  

I solved it by commenting the following lines:

```diff
--- a/quantum/pointing_device/pointing_device_auto_mouse.c
+++ b/quantum/pointing_device/pointing_device_auto_mouse.c
@@ -397,8 +397,8 @@ bool process_auto_mouse(uint16_t keycode, keyrecord_t* record) {
                 break;
             }
         // MT(kc) only skip on hold
-        case QK_MOD_TAP ... QK_MOD_TAP_MAX:
-            if (!record->tap.count) break;
+        /*case QK_MOD_TAP ... QK_MOD_TAP_MAX:*/
+        /*    if (!record->tap.count) break;*/
 #    endif
         // QK_MODS goes to default
         default:
```

closed 2025-02-25 by caut110