qmk/qmk_firmware #24959: Binary file Flash usage problem

Antecer, 2025-02-27:
### Issue Description

I found that the compiled bss always occupies all the unused ram, which causes me to be unable to tell the actual ram occupancy, how can I solve this problem?

mcu flash is： rom_256k, ram_96k

```
Size after:
text   data   bss    dec    hex      rom    ram    filename
62760  12872  85432  161064 0x27528  75632  98304  .build/debug.elf
```

tzarc, 2025-02-27:
Largely irrelevant, it’s fully used because of the memory allocator. You shouldn’t need to care.

sigprof, 2025-02-27:
If you want to estimate how much RAM is free, you can use `arm-none-eabi-size --format=sysv .build/$FILE.elf` and look at the size of the `.heap` section — that section occupies all unused RAM in the RAM region which is used for the heap. Actual usage of `malloc()` in QMK is rather rare (some debouncing algorithms allocate a block once at startup, and there is some dynamic allocation in Quantum Painter with `QUANTUM_PAINTER_LOAD_FONTS_TO_RAM` and in LVGL), therefore most of the heap area can be considered as available.

However, many MCUs actually have multiple RAM blocks which are not contiguous in the address space, and free space in those non-default RAM blocks may be harder to account for.  E.g., RP2040 mainly uses SRAM banks 0…3 as a single 256K block, and any memory in that block which was not allocated otherwise by the linker will be assigned to the `.heap` section; but there are also banks 4 and 5, which are used for ChibiOS stacks and other per-core data (bank 4 for core 0, and bank 5 for core 1); the usage of these blocks is less obvious in the `arm-none-eabi-size --format=sysv` output (you could see each section, but the memory block assignment may not be obvious, and multiple sections may be assigned to the same memory block).

Antecer, 2025-02-28:
> If you want to estimate how much RAM is free, you can use `arm-none-eabi-size --format=sysv .build/$FILE.elf` and look at the size of the `.heap` section — that section occupies all unused RAM in the RAM region which is used for the heap. Actual usage of `malloc()` in QMK is rather rare (some debouncing algorithms allocate a block once at startup, 和 there is some dynamic allocation in Quantum Painter with `QUANTUM_PAINTER_LOAD_FONTS_TO_RAM` and in LVGL), therefore most of the heap area can be considered as available.
> 
> However, many MCUs actually have multiple RAM blocks which are not contiguous in the address space, 和 free space in those non-default RAM blocks may be harder to account for. E.g., RP2040 mainly uses SRAM banks 0…3 as a single 256K block, 和 any memory in that block which was not allocated otherwise by the linker will be assigned to the `.heap` section; but there are also banks 4 and 5, which are used for ChibiOS stacks and other per-core data (bank 4 for core 0, 和 bank 5 for core 1); the usage of these blocks is less obvious in the `arm-none-eabi-size --format=sysv` output (you could see each section, but the memory block assignment may not be obvious, 和 multiple sections may be assigned to the same memory block).

Thanks, this is exactly what I need!

closed 2025-02-28 by Antecer