qmk/qmk_firmware #24966: [Bug] Compilation stops because of multiple target patterns after 128 git's error code

LemeRus, 2025-02-28:
### Describe the Bug

I tried to compile my firmware with `qmk -v compile -kb elcantorhs/42 -km default`, but see errors
```
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
builddefs/build_keyboard.mk:291: *** multiple target patterns.  Stop.
```
My qmk_firmware is placed in venv and was broken after changing mount point of partition. I reinstalled qmk, but it returns me an error (before changing of partition it was working normally, without issues)

Any ideas to fix this?

### Keyboard Used

elcantorhs/42

### Link to product page (if applicable)

https://github.com/azhizhinov/ELCANTORHS

### Operating System

Linux (Linux Mint 22)

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /mnt/870_EVO/Repos/qmk_venv/qmk_firmware
Ψ Detected Linux (Linux Mint 22).
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.28.0
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 7.3.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84429)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e4)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2025-02-28:
Use git, don’t download the zip file.

LemeRus, 2025-03-01:
> Use git, don’t download the zip file.

I installed qmk with `pip -m install qmk && qmk setup`. I didn't download zip file

tzarc, 2025-03-01:
My apologies then.  99.9% of the time we see that error it’s because the user has downloaded the zip and expected it to work.

There’s at least _some_ issue with the QMK CLI determining which version is in use in your environment.  If you’re familiar with Python debugging, that line in the makefile shows you which command it’s executing.

LemeRus, 2025-03-01:
> There’s at least _some_ issue with the QMK CLI determining which version is in use in your environment. If you’re familiar with Python debugging, that line in the makefile shows you which command it’s executing.

Unfortunately I'm not familar with python debugging and need help with this

elpekenin, 2025-03-01:
The line casuing error is `$(shell $(QMK_BIN) generate-version-h $(VERSION_H_FLAGS) -q -o $(INTERMEDIATE_OUTPUT)/src/version.h)`

This makes me think you are (no idea why/how) getting a Python exception when QMK tries to generate a `.h` file that contains some version markers. 

If you run the command yourself on a terminal with `qmk generate-version-h -q`, does it throw an error?

PS: Help would be faster if you joined discord and asked there

LemeRus, 2025-03-02:
> `qmk generate-version-h -q`

Thank you very much, this command has more verbose output, this helps.

Final small question)
What makes `git --add safe.directory`? What is safe.directory?

elpekenin, 2025-03-04:
I think that's mostly for CI runs, as otherwise the repository on the ephemeral runner is marked as not trusted and some thing dont work... Which would not matter here

Which output do you see?

tzarc, 2025-03-04:
> I think that's mostly for CI runs, as otherwise the repository on the ephemeral runner is marked as not trusted and some thing dont work... Which would not matter here
> 
> Which output do you see?

Irrelevant. It’s a `git` thing when dealing with files owned by someone else: https://git-scm.com/docs/git-config#Documentation/git-config.txt-safedirectory

closed 2025-03-07 by LemeRus

CDRXavier, 2026-01-02:
Can confirm with my setup.
Errors of 
```bash
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
builddefs/build_keyboard.mk:62: *** multiple target patterns.  Stop.
Make finished with errors
gmake: *** [Makefile:392: framework/numpad:default] Error 1
```
are exact, and the `SKIP_GIT=yes` fixes the issue.
```bash
xavier@LAPTOP-3010014:/mnt/Volume_A1/qmk_firmware$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /mnt/Volume_A1/qmk_firmware
Ψ Detected Linux (Ubuntu 24.04.3 LTS).
Ψ Git branch: fl16-bisect
Ψ Repo version: v0.3.0
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.2.1
Ψ Found avr-gcc version 7.3.0
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2023-04-15 13:48:04 +0000 --  (11edb16)
Ψ - lib/chibios-contrib: 2023-07-17 11:39:05 +0200 --  (da78eb3)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```