qmk/qmk_firmware #24969: [Bug] No rule to make target 'drivers/ws2812.c'

Speed1, 2025-03-01:
### Describe the Bug

I receive the following error when I try to compile with the latest update:

```
qmk compile -kb keycapsss/3w6_2040 -km default
gmake: *** No rule to make target 'drivers/ws2812.c', needed by '.build/obj_keycapsss_3w6_2040_speed1/ws2812.o'.  Stop.
```

Did I miss something?

### Keyboard Used

keycapsss/3w6_2040

### Link to product page (if applicable)

https://keycapsss.com/diy-kits/191/3w6-rp2040-split-keyboard-kit

### Operating System

MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /Users/speed/src/github/qmk_firmware
Ψ Detected macOS 15.3.1 (Apple Silicon).
Ψ Userspace enabled: False
Ψ Git branch: master
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.5.0
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 8.5.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

elpekenin, 2025-03-01:
Hard to help when doctor shows no information about your local clone of the code because you dont have the official repository set as `upstream`.

All i can say is that the code for the board ([3w6_2040](https://github.com/qmk/qmk_firmware/tree/master/keyboards/keycapsss/3w6_2040)) hasn't changed for almost a year, and it builds fine on my machine.

The fact that it is trying to compile `drivers/ws2812.c` is weird because that file doesnt work for RP2040, and neither does the code in the repository try and use it... Have you messed up with the keyboard/keymap makefiles?

Speed1, 2025-03-01:
Hi @elpekenin 

thanks for your response and the info that its working on your end. 
I have deleted the local repo and recloned it from my repo and now its working. Don't know what the problem was but issue can be closed.

Thanks

closed 2025-03-01 by Speed1

blefevre, 2025-04-01:
For anyone coming across this issue in the future it can be fixed by running `qmk clean` to delete old build objects in `.build/`.