qmk/qmk_firmware #24985: [Bug] Numpad comma doesn't work

pshabunia, 2025-03-04:
### Describe the Bug

Numpad comma doesn't work as expected. Keycodes `KC_KP_COMMA` or `KC_PCMM` have no effect.

When tested via https://config.qmk.fm/#/test it reads

`KEY-DOWN - QMK:         Event key: Unidentified Code: NumpadComma   KeyCode: 0`

### Keyboard Used

handwired/dactyl_manuform/5x6

### Operating System

MacOS Monterey 12.7.6

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /Users/pshabunia/Workspace/qmk_firmware
Ψ Detected macOS 12.7.6 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: pshabunia_dm_5x6
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 9.4.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [x] Karabiner (macOS)

sigprof, 2025-03-04:
The table at https://docs.qmk.fm/keycodes shows that `KC_KP_COMMA` works only on Linux.  This is a common issue with keys which are not present on standard 104/105-key keyboards — their support is often sparse.

pshabunia, 2025-03-05:
oh, I see now. I'm using the dark theme and I just couldn't see gray checkboxes on the black background. Also the table headers are fixed at the top of the page so it's not obvious what the checkboxes would mean.  

Totally my bad, but now I with the table used a floating header and/or colorful icons in the three rightmost columns

closed 2025-03-05 by pshabunia