qmk/qmk_firmware #24986: [Bug]

pshabunia, 2025-03-04:
### Describe the Bug

QMK doctor was failing on arm-none-eabi-gcc.
```
☒ Failed to compile a simple program with arm-none-eabi-gcc, return code 1
☒ Command: arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb -mno-thumb-interwork --specs=nosys.specs --specs=nano.specs -x c -o /var/folders/_0/f06cvz952575zgxdr5d0c_7c0000gn/T/tmpexopx9jl/test.elf -
```
When executed directly in terminal, the command returns `arm-none-eabi-gcc: fatal error: cannot read spec file 'nosys.specs': No such file or directory`

The problem seemed to be caused by faulty lib. Replacing it with newer version didn't help. I found a similar issue on raspberrypi forum and people reported success after replacing `arm-none-eabi-gcc` with `gcc-arm-embedded`. I tried and it worked for me.

Is this a bug or something was funky with my installation? I'm looking for peer review from the community.

---

faulty QMK doctor output:
```
ace enabled: False
Ψ Git branch: pshabunia_dm_5x6
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.2.0
☒ Failed to compile a simple program with arm-none-eabi-gcc, return code 1
☒ Command: arm-none-eabi-gcc -mcpu=cortex-m0 -mthumb -mno-thumb-interwork --specs=nosys.specs --specs=nano.specs -x c -o /var/folders/_0/f06cvz952575zgxdr5d0c_7c0000gn/T/tmpexopx9jl/test.elf -
Ψ Found avr-gcc version 9.4.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ Major problems detected, please fix these problems before proceeding.
```

output after corrections:
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /Users/pshabunia/Workspace/qmk_firmware
Ψ Detected macOS 12.7.6 (Intel).
Ψ Userspace enabled: False
Ψ Git branch: pshabunia_dm_5x6
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 9.4.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 0.7.2
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b973)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f)
Ψ QMK is ready to go, but minor problems were found
```

### Keyboard Used

handwired/dactyl_manuform/5x6

### Operating System

MacOS 12.7.6

### qmk doctor Output

```text
provided two examples above
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [x] Karabiner (macOS)

tzarc, 2025-03-04:
Happens often enough to have created #24637. You had the wrong toolchain.

pshabunia, 2025-03-05:
thank you for the info, Nick!