qmk/qmk_firmware #25006: [Bug] "Customizable key repress" cannot work well

tamura55, 2025-03-12:
### Describe the Bug

I want to implement the codes described in the following URL.
https://docs.qmk.fm/features/combo#customizable-key-repress

When combo F+G activation, registers Alt and presses Tab correctly.
And after released G, Alt is still registered. (so it is also OK)
But, when pressing G again, window is not switched.
Please let me know the reason.

Note: when keeping F and presses KC_TAB key (the other key, just assign KC_TAB), window is switched.
So I think there are some trouble in process_combo_key_repress().
(Of course, COMBO_PROCESS_KEY_REPRESS is defined)

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

waffle87, 2025-03-13:
Please attach your code or a link to your repository where it can be found. Paraphrasing makes it difficult to diagnose the issue.

tamura55, 2025-03-14:
Thank you for your comment!
Actually I found that QMK 0.22.14
does not support COMBO_PROCESS_KEY_REPRESS function.
So I will close this issue.

closed 2025-03-14 by tamura55