qmk/qmk_firmware #25008: [Bug] Quantum painter images dissapearing

CaffeinatedOpe, 2025-03-12:
### Describe the Bug

I'm using an SSD1306 screen with a KB2040 over i2c, and I've got a strange issue of the images beginning to disappear immediately after they are shown. important part of keymap.c is below, and a video showing the issue is attached.
`#include QMK_KEYBOARD_H
#include <qp.h>

static painter_device_t display;
void keyboard_post_init_kb(void) {
	display = qp_sh1106_make_i2c_device(128, 32, 0x3c);
	qp_init(display, QP_ROTATION_0);   // Initialise the display
	qp_circle(display, 64, 16, 4, 255, 255, 255, true);
	qp_flush(display);
}` 

https://github.com/user-attachments/assets/0fda0e53-3a91-4122-b7ee-3b5077db7723

 This issue is only present when using quantum painter, and is not present when using the older oled system. I do have serial split configured, but only one "half" is present.

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

tzarc, 2025-03-12:
Post a link to your keymap repo — can confirm that QP works perfectly fine on OLED test boards of mine.

CaffeinatedOpe, 2025-03-12:
https://github.com/CaffeinatedOpe/corneography is the repo, it wasn't properly forked as I didn't intend for it to be shared.

tzarc, 2025-03-12:
Unfortunately I can't do any diffs without history and if I'm honest I don't particularly feel like sinking time into downloading a copy and performing diffs manually locally, especially when there's no real indication of what revision it was forked at.

I'd suggest using the normal git workflow and keeping things sync'ed and public -- it's much too difficult to perform any diagnosis with such uncertainty.

CaffeinatedOpe, 2025-03-12:
Even if I had forked it, there wouldn't be much for diffs. The single keyboard profile in the repo was made from scratch, and is little more than a test bed for the oled screen

tzarc, 2025-03-12:
No problem; I'd suggest running through [debugging docs](https://docs.qmk.fm/faq_debug) and turning on [QUANTUM_PAINTER_DEBUG](https://docs.qmk.fm/quantum_painter#quantum-painter-config) -- perhaps something jumps out.

sigprof, 2025-03-13:
You have `"oled": true` in `keyboard.json` — probably the old OLED feature still tries to drive the display and causes some conflicts.

CaffeinatedOpe, 2025-03-13:
Removing that fixed the issue right up, thank you @sigprof !

closed 2025-03-13 by CaffeinatedOpe