qmk/qmk_firmware #25017: [Bug] is_caps_word_on() correct value not available when using tap dance

mhalano, 2025-03-15:
### Describe the Bug

My goal is to use tap dance to enable caps word with one key press and caps lock with two key press.
Caps lock is working as should, but I can't turn off caps word after enabling it. The relevant code on my keymap.c:
```c
void on_tap(tap_dance_state_t *state,  void *user_data) {

    switch (state->count) {
        case 1:
            if (host_keyboard_led_state().caps_lock) {
                tap_code16(KC_CAPS);
                break;
            }

            caps_word_toggle();
            break;

        case 2:
            tap_code16(KC_CAPS);
            break;
    }
}
```
I checked using if and SEND_STRING, and when the function on_tap is called, the `is_caps_word_on()` value is false, even if it's already enabled.
I was able to fix that when I created a variable outside the function on_tap and set the proper value inside:
```c
int caps_word_status;
void on_tap(tap_dance_state_t *state,  void *user_data) {
    switch (state->count) {
        case 1:
            if (host_keyboard_led_state().caps_lock) {
                tap_code16(KC_CAPS);
                break;
            }

            if (is_caps_word_on() || caps_word_status) {
                caps_word_off();
                caps_word_status = false;
                break;
            } else {
                caps_word_on();
                caps_word_status = true;
                break;
            }
            caps_word_toggle();
            break;

        case 2:
            tap_code16(KC_CAPS);
            break;
    }
}
```
In this case almost worked because if I exit caps word without pressing a key, like typing <space>, I had to press caps twice to enable, since the state of the variable is still true, but the caps word already was disabled.
The idea is to get the correct value for `is_caps_word_on()` inside the function used with tap dance.

### Keyboard Used

xbows/nature

### Operating System

Linux

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: /home/mhalano/code/qmk_firmware
Ψ Detected Linux (Ubuntu Plucky Puffin (development branch)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /home/mhalano/code/qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: master
Ψ Repo version: 0.28.0
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest master: 2025-03-13 15:10:54 -0300 (a9b854e843) -- Some changes because of the rebase feat not using CAPS word anymore
Ψ - Latest upstream/master: 2025-03-13 21:04:32 +0000 (65aef7be95) -- [Keyboard] Add Sofle Choc Prog from keebart (#24974)
Ψ - Latest upstream/develop: 2025-03-11 09:43:13 +0000 (3306d78bbd) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2025-03-11 09:42:33 +0000 (84f9b6dd3e) -- Add Mechboards Sofle (#25004)
Ψ - Common ancestor with upstream/develop: 2025-03-11 09:42:33 +0000 (84f9b6dd3e) -- Add Mechboards Sofle (#25004)
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.2.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.2.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.1
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.11
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

mhalano, 2025-03-15:
To help put in context the problem, I have another function used to control the LEDs and `is_caps_word_on()` returns the correct value:
```c
bool rgb_matrix_indicators_advanced_user(uint8_t led_min, uint8_t led_max) {
    for (uint8_t i = led_min; i < led_max; i++) {
        if (host_keyboard_led_state().caps_lock) {
            rgb_matrix_set_color(45, RGB_BLUE);
        } else if (is_caps_word_on()) {
            rgb_matrix_set_color(45, RGB_YELLOW);
        }
        rgb_matrix_set_color(i, RGB_WHITE);
    }
    return true;
}
```

zvecr, 2025-03-16:
As discussed on Discord, this is not a bug as the default behaviour is for tap dance to disable caps word.

https://github.com/qmk/qmk_firmware/blob/65aef7be95025a8a26fa8fe0c4af3ce0d36f47f4/quantum/process_keycode/process_caps_word.c#L254-L272

`process_caps_word` gets called before `process_tap_dance`, where `process_caps_word` will run `caps_word_press_user` to see if the current keypress should disable. This happens shortly before your tap dance "on_tap" function is called, resulting in `is_caps_word_on` returning false.

Overriding this function with one what returns true for tap dances, while implementing the `on_tap` logic within this issue, will allow you to mutate the caps word enabled state without it getting turned off automatically.

closed 2025-03-16 by zvecr

mhalano, 2025-03-16:
What could go wrong if we put an if condition before the switch, or another switch statement, in the function `caps_word_press_user()` for all `TD(*)` keycodes?