qmk/qmk_firmware #25019: [Bug] host_keyboard_led_state() not available inside rgb_matrix_indicators_user()

mhalano, 2025-03-15:
### Describe the Bug

I can't check if caps lock is enabled or not using `host_keyboard_led_state().caps_lock` when I'm inside the function `rgb_matrix_indicators_user()`.
When I'm using `rgb_matrix_indicators_advanced_user()` instead, it did work, but I don't care about any input in the function in this case.

Here is my code:
```c
bool rgb_matrix_indicators_advanced_user(uint8_t led_min, uint8_t led_max) {
    if (host_keyboard_led_state().caps_lock) {
        rgb_matrix_set_color(45, RGB_BLUE);
    } else if (is_caps_word_on()) {
        rgb_matrix_set_color(45, RGB_YELLOW);
    }
    return true;
}
```
If I replace the function it stops working.

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2025-03-15:
The various template fields you have ignored are not really optional when filing a bug.

Please fill these in.

elpekenin, 2025-03-15:
Please, dont open "bug" reports eagerly, first try and seek for support on discord (you are already in the server too), where i would have said the following
1) You didn't post the whole picture (ie: a link to your repo to see the all the code)
2) Taking a look at your fork i see that there is this: https://github.com/mhalano/qmk_firmware/blob/dd0f4c80446f9fc5582b30c7aaf6c699a0764d0c/keyboards/xbows/nature/nature.c#L143-L145
This is: whatever you do on `_user` is overriden by the `_kb` counterpart because you are doing `return true` (meaning: im fine with `_kb` logic still running), should `return false` instead

Im guessing that `indicators_advanced_user` dodged the bullet because it gets to draw after `indicators_kb`

mhalano, 2025-03-16:
Here it is my repository: https://github.com/mhalano/qmk_firmware/tree/master/keyboards/xbows/nature/keymaps/mhalano

closed 2025-03-16 by mhalano