qmk/qmk_firmware #25024: Keyboard Keychron C3 Pro firmware seems not working properly

DiamondGo, 2025-03-16:
### Issue Description

Keychron c3 pro support was added on Jan 29, 2025 by #24874 

After flashing this version, the mode switch LEDs are off and Fn+Caps can not switch between Mac/Win modes.
Also the keyboard is no longer recognized by VIA after flashing this firmware.

The backlighting is working properly.

Not sure if this is a bug or not. I am new to QMK and hopefully someone could help take a look, or tell me where I should look into.

Great thanks.

adophoxia, 2025-03-18:
What variant of the board do you have exactly, for context?

iamdanielv, 2025-03-21:
VIA won't work since it is not enabled in the code in the repo.

Fn+Caps should work to switch between Mac and Windows modes.

Have you tried using: https://config.qmk.fm/#/test
to see what keys are being reported?

Try pushing left alt and see what key code it reports, then push Fn+Caps and see if it changes.

Hope that helps!

iamdanielv, 2025-03-21:
Forgot to mention,
If you want to enable VIA, you can add a rules.mk file in your keymap directory and put "VIA_ENABLE = yes" in the content.

DiamondGo, 2025-03-24:
> What variant of the board do you have exactly, for context?

It's ansi red.

DiamondGo, 2025-03-24:
> Forgot to mention, If you want to enable VIA, you can add a rules.mk file in your keymap directory and put "VIA_ENABLE = yes" in the content.

> VIA won't work since it is not enabled in the code in the repo.
>
> Fn+Caps should work to switch between Mac and Windows modes.
>
> Have you tried using: https://config.qmk.fm/#/test to see what keys are [...]

Thanks.

By adding a rules.mk with `VIA_ENABLE = yes` , now VIA can recognize the keyboard.

But Fn+Caps still not working.

On the test page, left alt is reported as AltLeft, after Fn+Caps it's reported as MetaLeft. Does it mean the Fn+Caps is changing the layout but the Mac/Win leds are not working?

adophoxia, 2025-03-24:
> > Forgot to mention, If you want to enable VIA, you can add a rules.mk file in your keymap directory and put "VIA_ENABLE = yes" in the content.
> 
> > VIA won't work since it is not enabled in the code in the repo.
> > Fn+Caps should work to switch between Mac and Windows modes.
> > Have you tried using: https://config.qmk.fm/#/test to see what keys are being reported?
> > Try pushing left alt and see what key code it reports, then push Fn+Caps and see if it changes.
> > Hope that helps!
> 
> Thanks.
> 
> By adding a rules.mk with `VIA_ENABLE = yes` , now VIA can recognize the keyboard.
> 
> On the test page, left alt is reported as AltLeft, after Fn+Caps it's reported as MetaLeft. Does it mean the Fn+Caps is changing the layout but the Mac/Win leds are not working?

Currently changed the firmware to make it work properly over in https://github.com/qmk/qmk_firmware/pull/25049.

iamdanielv, 2025-03-24:
> Thanks.
>
> By adding a rules.mk with `VIA_ENABLE = yes` , now VIA can recognize the keyboard.
>
> On the test page, left alt is reported as AltLeft, after Fn+Caps it's reported as MetaLeft. Does it mean [...]

Yes, that's right. I don't use the MAC layers and repurposed the LEDs for other things. I have the same ANSI Red version.

iamdanielv, 2025-05-16:
Can this issue be closed?

closed 2025-09-14 by github-actions[bot]