qmk/qmk_firmware #25106: [Bug] Oneshot locked modifier is not disabled on the next modifier key press after toggling

denis-savran, 2025-04-07:
### Describe the Bug

If you toggle a oneshot locked modifier and then press the modifier key again to disable it, the modifier remains active in an OS until the next key press on a keyboard.

Steps to reproduce:

1. Toggle `Shift` by pressing it two times.
2. Press `Shift` one time.

Actual behavior:

The `Shift` modifier is not disabled right after pressing the modifier key.
It remains active in an OS.

To disable it in the OS you have to press some key on a keyboard (the modifier is not applied), trigger a hold event for the modifier key or restart the keyboard.

Expected behavior:

The `Shift` modifier is disabled right after pressing the modifier key.

This bug breaks the following usage scenario:

1. Select the first item using a mouse.
2. Toggle `Shift` by pressing it two times.
3. Click on the second item to select several items.
4. Disable `Shift` by pressing it one time.
5. Perform some action with the selection.

### Keyboard Used

moonlander

### Link to product page (if applicable)

https://www.zsa.io/moonlander

### Operating System

Ubuntu 22.04

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2025-04-08:
I'm not able to reproduce this issue, at all.  Unless I'm pressing it very fast.  Eg, all of this within tapping term of each press.  And if that is the case, then it's not that it's getting stuck. 

It might be a good idea to define `ACTION_DEBUG` in your config, and enable debug print.  This should print the actions to the console, so you can get a clearer idea of what exactly is happening. (eg `debug_enable = true` in `keyboard_pre_init_user`)

denis-savran, 2025-04-08:
Added the following changes to enable debug:

`rules.mk`
```diff
-CONSOLE_ENABLE = no
+CONSOLE_ENABLE = yes
```

`config.h`
```diff
+#define ACTION_DEBUG
```

`keymap.c`
```diff
+void keyboard_post_init_user(void) {
+    debug_enable=true;
+    debug_keyboard=true;
+}
```

Added a missing newline character:

`action.c`
```diff
-                                ac_dprintf("MODS_TAP: Toggling oneshot");
+                                ac_dprintf("MODS_TAP: Toggling oneshot\n");
```

Then flashed a new firmware, started `qmk console` and performed the following actions:
1) Pressed and immediately released `Shift`.
2) Pressed and immediately released `Shift` to lock the modifier.
3) Waited for 5 seconds.
4) Pressed and released `Shift` to unlock the modifier.

<details>
<summary>Log before the fix</summary>

```
Ψ Console Connected: ZSA Technology Labs Moonlander Mark I (3297:1969:1)
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1: ---- action_exec: start -----
ZSA Technology Labs:Moonlander Mark I:1: EVENT: 0300d(12641)
ZSA Technology Labs:Moonlander Mark I:1: Tapping: Start(Press tap key).
ZSA Technology Labs:Moonlander Mark I:1: TAPPING_KEY=0300d(12641):0
ZSA Technology Labs:Moonlander Mark I:1: processed: 0300d(12641):0
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1: ---- action_exec: start -----
ZSA Technology Labs:Moonlander Mark I:1: EVENT: 0300u(12700)
ZSA Technology Labs:Moonlander Mark I:1: Tapping: First tap(0->1).
ZSA Technology Labs:Moonlander Mark I:1: TAPPING_KEY=0300d(12641):1
ZSA Technology Labs:Moonlander Mark I:1: ACTION: ACT_LMODS_TAP[2:00] layer_state: 00000000(0) default_layer_state: 00000001(0)
ZSA Technology Labs:Moonlander Mark I:1: MODS_TAP: Oneshot: start
ZSA Technology Labs:Moonlander Mark I:1: waiting_buffer_enq: { [1]=0300u(12700):1 }
ZSA Technology Labs:Moonlander Mark I:1: ---- action_exec: process waiting_buffer -----
ZSA Technology Labs:Moonlander Mark I:1: Tapping: Tap release(1)
ZSA Technology Labs:Moonlander Mark I:1: ACTION: ACT_LMODS_TAP[2:00] layer_state: 00000000(0) default_layer_state: 00000001(0)
ZSA Technology Labs:Moonlander Mark I:1: TAPPING_KEY=0300u(12700):1
ZSA Technology Labs:Moonlander Mark I:1: processed: waiting_buffer[1] =0300u(12700):1
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1:
ZSA Technology Labs:Moonlander Mark I:1: ---- action_exec: start -----
ZSA Technology Labs:Moonlander Mark I:1: EVENT: 0300d(12777)
ZSA Technology Labs:Moonlander Mark I:1: Tapping: Tap press(2)
ZSA Technology Labs:Moonlander Mark I:1: ACTION: ACT_LMODS_TAP[2:00] layer_state: 00000000(0) default_layer_state: 00000001(0)
ZSA Technology Labs:Moonlander Mark I:1: MODS_TAP: Toggling oneshot
ZSA Technology Labs:Moonlander Mark I:1: keyboard_report: 02 | 00 00 00 00 00 00
ZSA Technology Labs:Moonlander Mark I:1: TAPPING_KEY=0300d(12777):2
[... 141 lines cut ...]
+ZSA Technology Labs:Moonlander Mark I:1: Tapping: End(Timeout after releasing last tap): 0000u(40178)
 ZSA Technology Labs:Moonlander Mark I:1: TAPPING_KEY=0000u(0):0
```

Your suggested change also resolves the problem:
```diff
 #        if defined(ONESHOT_TAP_TOGGLE) && ONESHOT_TAP_TOGGLE > 1
                             } else if (tap_count == 1 && (mods & get_mods())) {
-                                unregister_mods(mods);
                                 del_oneshot_mods(mods);
+                                unregister_mods(mods);
                                 del_oneshot_locked_mods(mods);
 #        endif
```

At first I resolved the issue by adding `send_keyboard_report` call:
```diff
                                 unregister_mods(mods);
                                 del_oneshot_mods(mods);
                                 del_oneshot_locked_mods(mods);
+                                send_keyboard_report();
 #        endif
```

But then found out that it is already called in `unregister_mods` function.

Jari27, 2025-04-21:
FYI, I faced a very similar issue:

https://github.com/qmk/qmk_firmware/pull/24925