qmk/qmk_firmware #25150: [Bug] WMIC missing on Windows 11 installation

molohov, 2025-04-14:
### Describe the Bug

Compilation of firmware is fine, but recently my work machine started to behave like this:

```
Flashing for bootloader: rp2040
Traceback (most recent call last):
  File "./qmk_firmware/util/uf2conv.py", line 372, in <module>
    main()
  File "./qmk_firmware/util/uf2conv.py", line 357, in main
    drives = get_drives()
             ^^^^^^^^^^^^
  File "./qmk_firmware/util/uf2conv.py", line 213, in get_drives
    r = subprocess.check_output(["wmic", "PATH", "Win32_LogicalDisk",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.12/subprocess.py", line 466, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.12/subprocess.py", line 548, in run
    with Popen(*popenargs, **kwargs) as process:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/QMK_MSYS/mingw64/lib/python3.12/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/QMK_MSYS/mingw64/lib/python3.12/subprocess.py", line 1538, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [WinError 2] The system cannot find the file specified
make[1]: *** [platforms/chibios/flash.mk:98: flash] Error 1
```
It appears this is a known issue as per messages on Discord, and there are workarounds such as [this](https://discord.com/channels/440868230475677696/867530222407778344/1332152471933681754), but I would like to open this issue to track a proper fix. 

### Keyboard Used

make ursula:molohov

### Operating System

Windows 11

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.6
Ψ QMK home: C:/Users/blong/Downloads/qmk_firmware
Ψ Detected Windows 11 (10.0.26100).
Ψ QMK MSYS version: 1.10.0
Ψ Userspace enabled: False
Ψ Git branch: molohov
Ψ Repo version: 0.25.7
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest molohov: 2025-04-14 10:58:46 -0700 (a02c426661) -- Update keymaps and combos
Ψ - Latest upstream/master: 2025-04-14 10:27:52 -0700 (80612c8ec0) -- Added Keyboard LumPy27 (#24967)
Ψ - Latest upstream/develop: 2024-06-07 13:26:23 +0000 (3de5e44f52) -- Merge remote-tracking branch 'origin/master' into develop
Ψ - Common ancestor with upstream/master: 2025-04-14 10:27:52 -0700 (80612c8ec0) -- Added Keyboard LumPy27 (#24967)
Ψ - Common ancestor with upstream/develop: 2024-06-07 13:26:23 +0000 (3de5e44f52) -- Merge remote-tracking branch 'origin/master' into develop
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 13.3.0
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.2.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 7.3-2024041
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84429)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e4)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2025-04-14:
This should probably be reported at https://github.com/microsoft/uf2, which is where `util/uf2conv.py` comes from.

closed 2025-04-14 by molohov

molohov, 2025-04-14:
A simple workaround is to just reinstall WMIC: https://techcommunity.microsoft.com/blog/windows-itpro-blog/how-to-install-wmic-feature-on-demand-on-windows-11/4189530