qmk/qmk_firmware #25220: Teensy 4.0 eeprom not save.

alcnonl, 2025-05-01:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I'm trying to permanently save data to EEPROM on my Teensy 4.0 device using QMK. The data appears to be saved, but after unplugging and plugging the device back in, the data is lost. Is there no way to save data permanently? How is EEPROM used on ARM-based microprocessors? I need to store persistent data somehow - is this possible or could it be possible?

PeterMortensen, 2025-05-03:
***How*** do you try to store data to EEPROM?

tzarc, 2025-08-02:
Teensy 4 has no driver for flash writing nor does it have a driver for any sort of true nor emulated EEPROM.

It also has no support for external peripherals.

This is an underlying issue with the low-level MCU support provided by ChibiOS-Contrib. QMK has nothing prioritised to address this.

closed 2025-08-02 by tzarc