qmk/qmk_firmware #25267: Help flashing QMK firmware to RS40 keyboard (APM32F103)

ma2kaz, 2025-05-16:
### Issue Description

Hello QMK team,

I purchased an RS40 keyboard from AliExpress, and I’m trying to flash QMK firmware to it.

After checking the PCB, I confirmed that it uses an APM32F103 microcontroller.

I found a custom keymap and board definition by Audite Marlow on GitHub:
https://github.com/AuditeMarlow/qmk_userspace/tree/main/keyboards/kj_modify/rs40

However, I'm not sure how to properly flash the firmware using this board/chip. I’ve tried various approaches, but none have been successful so far.

Could you advise me on how to proceed with flashing QMK to this board?  
If needed, I can provide error messages and more details.

Thank you in advance!

ma2kaz, 2025-05-17:
Environment
- OS: Windows 11
- QMK CLI Version: 1.1.6
- dfu-util Version: 0.11
- Keyboard Directory: ./qmk_firmware/keyboards/kj_modify/rs40/
- MCU: APM32F103CBT6 (clone of STM32F103)
- Bootloader: stm32-dfu

rules.mk
```#MCU = STM32F103
#MCU = STM32_F103_STM32DUINO
#BOOTLOADER = stm32duino
#BOOTLOADER = stm32-dfu
#BOOTLOADER = rp2040

#MCU = STM32F103
BOOTLOADER = stm32-dfu
#BOARD = GENERIC_STM32_F103XB
#CHIBIOS_BOARD = GENERIC_STM32_F103XB
#PLATFORM = chibios

VIA_ENABLE = yes

$(info [DEBUG] CHIBIOS_BOARD = $(CHIBIOS_BOARD))
$(info [DEBUG] MCU = $(MCU))
```

- keyboard.json (unchanged): processor field is set to "STM32F401" (left as is for now)
- keymap.c: Unchanged default keymap

Compilation Output
```$ qmk compile -kb kj_modify/rs40 -km default
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=kj_modify/rs40 KEYMAP=default KEYBOARD_FILESAFE=kj_modify_rs40 TARGET=kj_modify_rs40_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

[DEBUG] CHIBIOS_BOARD = 
[DEBUG] MCU = 
arm-none-eabi-gcc.exe (GCC) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   32464       0   32464    7ed0 kj_modify_rs40_default.bin

Compiling: quantum/via.c                                                                            [OK]
Linking: .build/kj_modify_rs40_default.elf                                                          [OK]
Creating binary load file for flashing: .build/kj_modify_rs40_default.bin                           [OK]
Creating load file for flashing: .build/kj_modify_rs40_default.hex                                  [OK]

Size after:
   text    data     bss     dec     hex filename
      0   32464       0   32464    7ed0 kj_modify_rs40_default.bin

Copying kj_modify_rs40_default.bin to qmk_firmware folder
```

Flashing Output via QMK
```$ qmk flash -kb kj_modify/rs40 -km default
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s flash KEYBOARD=kj_modify/rs40 KEYMAP=default KEYBOARD_FILESAFE=kj_modify_rs40 TARGET=kj_modify_rs40_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

[DEBUG] CHIBIOS_BOARD = 
[DEBUG] MCU = 
arm-none-eabi-gcc.exe (GCC) 13.3.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   32464       0   32464    7ed0 kj_modify_rs40_default.bin

Compiling: quantum/via.c                                                                            [OK]
Linking: .build/kj_modify_rs40_default.elf                                                          [OK]
Creating binary load file for flashing: .build/kj_modify_rs40_default.bin                           [OK]
Creating load file for flashing: .build/kj_modify_rs40_default.hex                                  [OK]

Size after:
   text    data     bss     dec     hex filename
      0   32464       0   32464    7ed0 kj_modify_rs40_default.bin

Copying kj_modify_rs40_default.bin to qmk_firmware folder                                           [OK]
Flashing for bootloader: stm32-dfu
Device has DFU interface, but has no DFU functional descriptor
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Device has DFU interface, but has no DFU functional descriptor
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
Device ID 1209:db42
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
error get_status: LIBUSB_ERROR_PIPE
make: *** [platforms/chibios/flash.mk:98: flash] エラー 74
```

Manual dfu-util Flash Attempt
```$ dfu-util -a 0 -d 0483:DF11 -s 0x8000000 -D ./kj_modify_rs40_default.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Device has DFU interface, but has no DFU functional descriptor
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
Device ID 1209:db42
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
error get_status: LIBUSB_ERROR_PIPE
```

Notes
- The board appears to enter DFU mode (VID:PID is 1209:db42), but dfu-util reports a missing functional descriptor and fails with LIBUSB_ERROR_PIPE.
- Tried both automatic (qmk flash) and manual (dfu-util) flashing methods.
- The board uses APM32F103CBT6 instead of official STM32F103.
- Also tested with the board definition located at ./qmk_firmware/lib/chibios-contrib/os/hal/boards/ST_STM32F0308_DISCOVERY, but encountered the same LIBUSB_ERROR_PIPE error during flashing.

tzarc, 2025-05-17:
Again, whatever you’re seeing isn’t the same board as what’s in QMK. Suggest contacting your vendor — we can’t do anything here.

closed 2025-05-17 by tzarc