qmk/qmk_firmware #25281: Can I use "make" to build all keymaps?

Antecer, 2025-05-20:
I found that when I only enter the make command without entering the keymapPath, the following error will appear. The build will not stop, but continue to build all keymaps in sequence, and `Making 'keyboard name' with keymap via ` will appear in sequence, but at this time the console will not output the c file build information, and the usage of flash will only be output after the build is completed.

```
$ make
Keyboard Firmware v1.0.1
☒ 'matrix_size'
Traceback (most recent call last):
  File "D:/QMK_MSYS/mingw64/lib/python3.12/site-packages/milc/milc.py", line 606, in __call__
    return self.__call__()
           ^^^^^^^^^^^^^^^
  File "D:/QMK_MSYS/mingw64/lib/python3.12/site-packages/milc/milc.py", line 611, in __call__
    return self._subcommand(self)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "H:/MyProjects/hall_effect/lib/python/qmk/decorators.py", line 27, in wrapper
    return func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^
  File "H:/MyProjects/hall_effect/lib/python/qmk/cli/list/layouts.py", line 21, in list_layouts
    info_data = info_json(cli.config.list_layouts.keyboard)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "H:/MyProjects/hall_effect/lib/python/qmk/info.py", line 165, in info_json
    info_data = _extract_led_config(info_data, str(keyboard))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "H:/MyProjects/hall_effect/lib/python/qmk/info.py", line 687, in _extract_led_config
    cols = info_data['matrix_size']['cols']
           ~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'matrix_size'
```

Are binaries built this way reliable?

tzarc, 2025-05-20:
Probably not. If you want to build everything, use `qmk mass-compile -km all`.

Antecer, 2025-05-20:
Thanks for your reply!

closed 2025-05-20 by Antecer

Antecer, 2025-05-20:
I just try it, then i get all failed:

[image]

tzarc, 2025-05-20:
Did you clean first? Does building any single one of them in isolation succeed?

Antecer, 2025-05-20:
I just fixed it.
It need clone git repository, if download zip from github, it will throw error "The folder is not a git repository"