qmk/qmk_firmware #25283: [Bug] can't compile splitkb/aurora/sofle_v2

Theoreticallyhugo, 2025-05-20:
### Describe the Bug

neither the default, nor [my own keymap](https://github.com/Theoreticallyhugo/qmk_firmware) of splitkb/aurora/sofle_v2 can be compiled on either of my two machines at the moment. both crash with the same errors which talk about something with encoders. when i rip out the encoders, i get different errors. i'll look more into how to write configs with current qmk, but would be really happy if someone could help me fix at least the default config as both that one and my own used to work.

in the meantime i'll also shoot splikb an email about this.

### Keyboard Used

splitkb/aurora/sofle_v2

### Link to product page (if applicable)

https://splitkb.com/collections/keyboard-kits/products/aurora-sofle-v2

### Operating System

Fedora 42 and MacOS

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.7
Ψ QMK home: /home/hugo/programming/qmk_firmware
Ψ Detected Linux (Fedora Linux 42 (KDE Plasma Desktop Edition)).
Ψ Userspace enabled: False
Ψ Git branch: master
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream".
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 14.1.0
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 14.2.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 8.0
Ψ Found dfu-programmer version 1.1.0
Ψ Found dfu-util version 0.11
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2024-12-15 07:44:56 +0000 --  (2365f84429)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e4)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2023-02-12 20:19:37 +0100 --  (a3398d8)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [x] Karabiner (macOS)

tzarc, 2025-05-20:
Post the actual error.

Theoreticallyhugo, 2025-05-20:
You're right... I don't know why it works now. I've tried this all multiple times in different configurations, with at least two clean installs i thought. It never worked, and now it does.

I'm very sorry and very thankful simultaneously.
cheers <3

closed 2025-05-20 by Theoreticallyhugo