qmk/qmk_firmware #25299: I am unable to compile a firmware

omzart, 2025-05-24:
### Issue Description

Hello

I am very new to this website and the firmware, I followed guides to compile a firmware for my keyboard, but I end up with errors and no firmware :(

Here the result: 

$ qmk compile -kb <icl03> -km default
bash: icl03: No such file or directory
[Ordinateur@Nzxt ~]$ qmk compile -kb designedbygg/ironclad/icl03 -km default
Ψ Compiling keymap with make -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=designedbygg/ironclad/icl03 KEYMAP=default KEYBOARD_FILESAFE=designedbygg_ironclad_icl03 TARGET=designedbygg_ironclad_icl03_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

☒ Not including data from file: keyboards/designedbygg/ironclad/icl03/keyboard.json
☒       development_board: 'sn32f240b_shared_matrix' is not one of ['promicro', 'elite_c', 'elite_pi', 'proton_c', 'kb2040', 'promicro_rp2040', 'blok', 'michi', 'bit_c_pro', 'stemcell', 'bluepill', 'blackpill_f401', 'blackpill_f411', 'bonsai_c4', 'helios', 'liatris', 'imera', 'svlinky']
☒ designedbygg/ironclad/icl03: No LAYOUTs defined! Need at least one layout defined in info.json.
☒ Not including data from file: keyboards/designedbygg/ironclad/icl03/keyboard.json
☒       development_board: 'sn32f240b_shared_matrix' is not one of ['promicro', 'elite_c', 'elite_pi', 'proton_c', 'kb2040', 'promicro_rp2040', 'blok', 'michi', 'bit_c_pro', 'stemcell', 'bluepill', 'blackpill_f401', 'blackpill_f411', 'bonsai_c4', 'helios', 'liatris', 'imera', 'svlinky']
☒ designedbygg/ironclad/icl03: No LAYOUTs defined! Need at least one layout defined in info.json.

 * No bootloader specified. Please set an appropriate bootloader in your rules.mk or info.json.        [ERRORS]

platforms/avr/bootloader.mk:144: *** Invalid BOOTLOADER. Arrêt.

I am using this github : 

https://github.com/SRGBmods/SonixQMK/tree/sn32_develop/keyboards/designedbygg/ironclad

Thanks and Regards

zvecr, 2025-05-24:
From our Discord bot:
> Support for Sonix-based keyboards (Womier, Redragon, Keychron etc.) is maintained in another fork, and is currently not part of upstream QMK. Please see below for help with these boards.
>
> GitHub: https://github.com/SonixQMK/
> Discord: https://discord.gg/q8VjhRgzRw

And:
> Support is maintained in another fork and is not part of QMK yet.
>
> Site: https://signalrgb.com/
> Git: https://gitlab.com/signalrgb/qmk_firmware
> Discord: https://discord.com/invite/signalrgb
> Docs: https://docs.signalrgb.com/qmk/qmk-introduction

closed 2025-05-24 by zvecr

omzart, 2025-05-24:
Ok thanks you i'll investigate this way