qmk/qmk_firmware #25310: [Bug] eeconfig_update_keymap now requires keymap_config_t pointer (QMK update regression)

jaeryongim, 2025-05-27:
### Describe the Bug

I pulled the latest version of the master today and then builds are not working properly.
 

error messages====================================================================

Ψ Compiling keymap with gmake -r -R -f builddefs/build_keyboard.mk -s KEYBOARD=totem KEYMAP=default KEYBOARD_FILESAFE=totem TARGET=totem_default VERBOSE=false COLOR=true SILENT=false QMK_BIN="qmk"

arm-none-eabi-gcc (Homebrew ARM GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/keymap_introspection.c                                                          In file included from quantum/keymap_introspection.c:9:
./keyboards/totem/keymaps/default/keymap.c: In function 'process_record_user':
./keyboards/totem/keymaps/default/keymap.c:104:53: error: passing argument 1 of 'eeconfig_update_keymap' makes pointer from integer without a cast [-Werror=int-conversion]
                 eeconfig_update_keymap(keymap_config.raw);
                                        ~~~~~~~~~~~~~^~~~
In file included from quantum/keycode_config.h:21,
                 from quantum/quantum.h:41,
                 from ./.build/obj_totem_default/src/default_keyboard.h:28,
                 from ./keyboards/totem/keymaps/default/keymap.c:1,
                 from quantum/keymap_introspection.c:9:
quantum/eeconfig.h:75:77: note: expected 'const keymap_config_t *' {aka 'const union keymap_config_t *'} but argument is of type 'uint16_t' {aka 'short unsigned int'}
 void                          eeconfig_update_keymap(const keymap_config_t *keymap_config) __attribute__((nonnull));
                                                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]

==========================================================================

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

sigprof, 2025-05-27:
That's an API change from #24356 (most `eeconfig_*` functions were changed that way).  Maybe it should have been better documented (https://docs.qmk.fm/ChangeLog/20250525 mentions only the PR name “Non-volatile memory data repository pattern”, and the association with eeconfig API changes is not obvious).

tzarc, 2025-05-27:
This would’ve been correctly modified already if the board was upstreamed into QMK actual. Suggest engaging the author of the board to fix it as per current QMK requirements — we can’t stay on top of 40,000+ forks.

closed 2025-05-27 by tzarc

jaeryongim, 2025-05-28:
@tzarc 
Oh, sorry. This is not a bug. i think I misunderstood your conversation a bit. 
I've only been using qmk for 4 days, so I'm a little new to looking for changes. 

I understood you to say at the end that not officially included in qmk, so you couldn't change it.

If there is such a change, I would appreciate if you could tell me how to find it or what to look for.

thx!

jaeryongim, 2025-05-28:
@tzarc  Thank you for your kind response. 
As for how to change it, I've seen and followed the changes on other keyboards, I was just wondering if the only way to do it is to look up the code if the some changes.

Again, thank you for your response.

tzarc, 2025-05-28:
In your case, probably the only way -- as #24356 keyboards in the repository have been migrated across, and end-users such as yourself would not really have any need to modify their code.