qmk/qmk_firmware #25314: Can't solve my RGB matrix

Gabriele-tomai00, 2025-05-28:
Hello, I'm having problem with my rgb matrix. In a file <keyboard_name>.c I put this code:
```
#ifdef RGBLIGHT_ENABLE
led_config_t g_led_config = { {
            {34,     33, 32, 31, 30, 29, 28},
            {22, 23, 24, 25, 26, 27, NO_LED},
            {21, 20, 19, 18, 17, 16, NO_LED},
            {9,      10, 11, 12, 13, 14, 15},
            {8,      7,   6,  5,  4,  3,  2},
            {NO_LED, NO_LED, NO_LED, NO_LED, 1, NO_LED, 0},

                                    {NO_LED, 64,     65, 66, 67, 68, 69, 70},
                                    {63,     62,     61, 60, 59, 58, 57, 56},
                                    {49,     50, 51, 52, 53, 54, 55, NO_LED},
                                    {48, 47, 46, 45, 44, 43, NO_LED, NO_LED},
                                    {37, 38, 39, 40, 41, 42, NO_LED, NO_LED},
                                    {36, 35, NO_LED, NO_LED, NO_LED, NO_LED, NO_LED, NO_LED}}, 
{
          // right part
          4, 4, 4, 4, 4, 4, 4, 
          4, 4, 4, 4, 4, 4,
          4, 4, 4, 4, 4, 4, 
          4, 4, 4, 4, 4, 4, 4,
          4, 4, 4, 4, 4, 4, 4 
                         4, 4,
          // left part
          4, 4, 4, 4, 4, 4, 4, 
       4, 4, 4, 4, 4, 4, 4, 4,
          4, 4, 4, 4, 4, 4, 4,
             4, 4, 4, 4, 4, 4,
             4, 4, 4, 4, 4, 4, 
             4, 4

     }
 };

#endif

```

But I got a very long list of errors during compilation:
```
Compiling: keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c                     In file included from quantum/quantum.h:32,
                 from keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:1:
quantum/rgblight/rgblight.h:325:22: error: expected identifier or '(' before 'void'
 void rgblight_enable(void);
                      ^~~~
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:5:1: error: unknown type name 'led_config_t'; did you mean 'debug_config_t'?
 led_config_t g_led_config = { {
 ^~~~~~~~~~~~
 debug_config_t
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:5:1: error: braces around scalar initializer [-Werror]
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:5:1: note: (near initialization for 'g_led_config')
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:13: error: braces around scalar initializer [-Werror]
             {34,     33, 32, 31, 30, 29, 28},
             ^
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:13: note: (near initialization for 'g_led_config')
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:22: error: excess elements in scalar initializer [-Werror]
             {34,     33, 32, 31, 30, 29, 28},
                      ^~
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:22: note: (near initialization for 'g_led_config')
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:26: error: excess elements in scalar initializer [-Werror]
             {34,     33, 32, 31, 30, 29, 28},
                          ^~
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:26: note: (near initialization for 'g_led_config')
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:30: error: excess elements in scalar initializer [-Werror]
             {34,     33, 32, 31, 30, 29, 28},
                              ^~
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:30: note: (near initialization for 'g_led_config')
keyboards/new_split_keyboard_gabriele/new_split_keyboard_gabriele.c:6:34: error: excess elements in scalar initializer [-Werror]
             {34,     33, 32, 31, 30, 29, 28},

....
```
This is the first keymap:
```
    [0] = LAYOUT(
      KC_GRV, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6,                                KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, 
      KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T,                                  KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS,          KC_MUTE,
      KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G,                                   KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, 
      KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_F1,                          KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, 
      MO(1), KC_LCTL, KC_LGUI, KC_LALT, KC_SPC, KC_ENT, KC_DEL,                   KC_SPC, MO(1), KC_LALT, KC_LGUI, KC_LEFT, KC_RIGHT,
                                MO(2), MO(1),                                      MO(1), KC_BSPC
    ),
```
I don't have problem without led. I'm building a custom split keyboard (serial connection) with RP2040

Thank you very much

zvecr, 2025-05-28:
```c
#ifdef RGBLIGHT_ENABLE
led_config_t g_led_config = { {
```

`RGBLIGHT` does not use `g_led_config`, only RGB Matrix does <https://docs.qmk.fm/features/rgb_matrix#common-configuration>.

yeroca, 2025-05-30:
Shouldn't it be rgb_config_t, rather than led_config_t?  The latter is for monochrome LEDs.