qmk/qmk_firmware #25374: [Bug] Installing QMK via `uv` gives error during installation of python dependencies

kartikay-bagla, 2025-06-15:
### Describe the Bug

I followed the [newbs setup guide](https://docs.qmk.fm/newbs_getting_started) and install qmk with `qmk tool install uv`. Followed the `qmk setup` process, which installed all dependencies.

Then when i did a test compile with `qmk compile -kb crkbd -km kb`, I got the following error:

```
Could not find module appdirs!
Would you like to install the required Python modules? [y/n] y
/home/kartikay/.local/share/uv/tools/qmk/bin/python: No module named pip
/home/kartikay/programming-linux/qmk_firmware_beekeeb/lib/python/qmk/decorators.py:20: UserWarning: cli._subcommand has been deprecated, please use cli.subcommand_name to get the subcommand name instead.
  if cli.config_source[cli._subcommand.__name__]['keyboard'] != 'argument':
/home/kartikay/programming-linux/qmk_firmware_beekeeb/lib/python/qmk/decorators.py:40: UserWarning: cli._subcommand has been deprecated, please use cli.subcommand_name to get the subcommand name instead.
  if cli.config_source[cli._subcommand.__name__]['keymap'] != 'argument':
Ψ Compiling keymap with gmake --jobs=1 crkbd/rev1:kb

Please run `/home/kartikay/.local/share/uv/tools/qmk/bin/python -m pip install -r /home/kartikay/programming-linux/qmk_firmware_beekeeb/requirements.txt` to install required python dependencies.
ERROR: Cannot run "qmk hello"!

 Please run qmk setup to install all the dependencies QMK requires.

gmake: *** [Makefile:396: crkbd/rev1:kb] Error 1
```

Now I believe this is due to qmk incorrectly assuming that pip is present in the build environment. The issue might be similar to [this](https://github.com/astral-sh/uv/issues/2725#issuecomment-2027254089) which was opened on uv's github.

To test this assumption, I ran `uv tool install --with pip qmk` and then ran the compile command again and was able to get it to work.

Potential solutions:
- add `pip` to dependencies for use as a uv tool
- update docs to show `uv tool install --with pip qmk` instead of `uv tool install qmk`

Note: I am using [this repository](https://github.com/kartikay-bagla/qmk_firmware_beekeeb) as my home.

### Keyboard Used

crkbd

### Operating System

Debian 12

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

drashna, 2025-07-06:
Shouldn't `sudo apt install -y git python3-pip` install pip, and prevent this issue from occurring (this is listed before the install command for pip or uv)

kartikay-bagla, 2025-07-06:
Apparently I did not RTFM closely.

This can be attributed to user error. Thanks for replying.