qmk/qmk_firmware #25425: [Bug] Weird issue with SPLIT_POINTING_ENABLE

anacierdem, 2025-06-28:
### Describe the Bug

I've recently moved to Fedora silver blue from Win10. I used to flash my keyboard with QMK toolbox and build it with MSYS.
This is my custom configuration branch: https://github.com/anacierdem/qmk_firmware/tree/custom-build

I'm pretty sure this build was working flawlessly when I last built it on Win10. I've built the firmware for the first time on Fedora today using `util/docker_build.sh` and mysteriously my keyboard stopped working on the right side (the one with pimoroni pointer), independent from which part is the master.

I've isolated it down to `SPLIT_POINTING_ENABLE`. Disabling it fixes the problem, but obviously pimoroni only works when master is right. Notice that my custom pimoroni RGB works even if the master is left.

I don't know what can cause a non-msys build can break things. In hopes for some new update fixing this I unfortunately rebased my branch on latest master so I currently don't know the working commit on Windows (I haven't retried building this latest on Windows either though, but I can have a look at the working setup by booting into my old setup)

I've currently disabled `SPLIT_POINTING_ENABLE` as I don't use the pointer much anyways. I'm very curious on why this might be happening though?

To summarize, enabling `SPLIT_POINTING_ENABLE`, disables right hand side (with pimoroni) independent of which one is the master.

### Keyboard Used

crkbrd/naci_rev

### Operating System

Bluefin

### qmk doctor Output

```text
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 1.1.8
Ψ QMK home: /qmk_firmware
Ψ Detected Linux (Debian GNU/Linux 11 (bullseye)).
⚠ Missing or outdated udev rules for 'atmel-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'kiibohd' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'stm32-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'apm32-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'gd32v-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'wb32-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'at32-dfu' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'bootloadhid' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbasploader' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'usbtinyisp' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'md-boot' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'caterina' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
⚠ Missing or outdated udev rules for 'hid-bootloader' boards. Run 'sudo cp /qmk_firmware/util/udev/50-qmk.rules /etc/udev/rules.d/'.
Ψ Userspace enabled: False
Ψ Git branch: custom-build
Ψ Repo version: 0.20.5
⚠ Git has unstashed/uncommitted changes.
Ψ - Latest custom-build: 2025-03-21 20:19:16 +0300 (1c4424a66e) -- Only enable HOLD_ON_OTHER_KEY_PRESS for shift key
Ψ - Latest upstream/master: 2025-06-28 11:29:57 -0500 (177e461d90) -- [Keyboard] Add togkey pad & pad_plus (#25217)
Ψ - Latest upstream/develop: None
Ψ - Common ancestor with upstream/master: 2025-06-28 11:29:57 -0500 (177e461d90) -- [Keyboard] Add togkey pad & pad_plus (#25217)
Ψ - Common ancestor with upstream/develop: None
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Successfully compiled using arm-none-eabi-gcc
Ψ Successfully tested arm-none-eabi-binutils using arm-none-eabi-size
Ψ Found avr-gcc version 8.3.0
Ψ Successfully compiled using avr-gcc
Ψ Successfully tested avr-binutils using avr-size
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-programmer version 0.6.1
Ψ Found dfu-util version 0.9
Ψ Submodules are up to date.
Ψ Submodule status:
Ψ - lib/chibios: 2025-05-15 08:25:11 +0000 --  (8bd61b8043)
Ψ - lib/chibios-contrib: 2025-01-08 21:03:31 +0100 --  (3ac181e4)
Ψ - lib/googletest: 2021-06-11 06:37:43 -0700 --  (e2239ee6)
Ψ - lib/lufa: 2022-08-26 12:09:55 +1000 --  (549b97320)
Ψ - lib/vusb: 2022-06-13 09:18:17 +1000 --  (819dbc1)
Ψ - lib/printf: 2022-06-29 23:59:58 +0300 --  (c2e3b4e)
Ψ - lib/pico-sdk: 2025-04-20 21:24:29 +1000 --  (d0c5cac)
Ψ - lib/lvgl: 2022-04-11 04:44:53 -0600 --  (e19410f8)
Ψ QMK is ready to go, but minor problems were found
```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

### Other keyboard-related software installed

N/A

drashna, 2025-07-06:
`NO_USB_STARTUP_CHECK` shouldn't be defined and may cause issues.  
Also you have ee_hands config enabled, but do you have this set up correctly on both halves?  If not, then it would prevent the split pointing code from properly initializing the pointing device.

anacierdem, 2025-07-06:
Thank you @drashna for having a look 🙏🏻

`NO_USB_STARTUP_CHECK` seemingly fixed my pre-logon issues. But I'll remove and test without that for some time b/c I was never sure if that helped or not.

I'm flashing with `:avrdude-split-left` & `:avrdude-split-right` respectively so I don't think it's related. Previously I was setting left&right from QMK toolkit UI but I suspect it does the same thing?

anacierdem, 2025-09-29:
`NO_USB_STARTUP_CHECK` indeed improved things. Pointer device now works fine, but `ENABLE_RGB_MATRIX_SOLID_REACTIVE_SIMPLE` still not working correctly. Let me try to figure out why that might be.