qmk/qmk_firmware #25445: [Bug] Mod-tap auto repeats key that is not pressed with two mod-taps

hadronized, 2025-07-02:
### Describe the Bug

Hello,

This is a pretty weird setup so I’ll try to explain as easy as possible. I noticed a weird behavior lately when using mod-tap with a high tapping term. I initially had a lower tapping term, but weird extra whitespaces, and thus realized something was really off by increasing the tapping term. I currently have this setting:

```c
#define TAPPING_TERM 2000
#define PERMISSIVE_HOLD
```

I have, on my Corne, assigned to two keys `LT(SYM, KC_SPC)` and `RSFT_T(KC_BSPC)` — note: I also tried it with other keys, the pattern is the same.

Now, the following sequences of keys:

1. Press `LT(SYM, KC_SPC)`.
2. Press `RSFT_T(KC_BSPC)`.
3. Release `LT(SYM, KC_SPC)`.
4. Wait until the end of the tapping term (around 2s).

Even though the space key is released, keeping pressing on the shift key outputs a repeated sequences of spaces. I had a look in `wev`, and it basically behaves as if the space key was pressed and not released, but gets released at the end of the tapping term.

I tried disabling flow tap, chordal hold, etc. etc. (as the setup above showed, none of that is enabled), and the problem still holds. The only way I have found to fix that issue is to remove the `LT` on the space key, which is not ideal to me.

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

hadronized, 2025-07-03:
I think it might be related to a similar issue as https://github.com/qmk/qmk_firmware/pull/17282

getreuer, 2026-11-07:
This bug sounds like yet another instance of what's described in https://github.com/qmk/qmk_firmware/issues/26559 and https://github.com/tmk/tmk_keyboard/issues/60.  If a key *release* is delayed because it is queued behind an unsettled tap-hold key, then depending on the kind of key and OS settings, it may be held artificially long enough to trigger auto-repeating. Of course, such an issue is more easily triggered with a large tapping term, like @hadronized described.

The fix is that the tapping engine should "forward" certain events to handling them immediately instead of processing in strict FIFO order. See also [the description of this PR](https://github.com/qmk/qmk_firmware/pull/26542) where I looked into a similar consideration of forwarding releases when using tap-hold keys with Auto Shift.

I'm endeavoring to round up tap-hold-related issues in https://github.com/qmk/qmk_firmware/issues/26548.