qmk/qmk_firmware #25451: [Bug] promicro_to_rp2040_ce converter doesnt know about the bottom row of pins on CE boards

rossw42, 2025-07-05:
### Describe the Bug

example pin 14 on a CE board - is C7 - 
QMK would not complie if I used this pin in a matrix
adding the definition of these pins to "promicro_to_rp2040_ce allowed for compiling.

I only did pin 14 - for completeness all the bottom row should be defined.

// Bottom row
#define C7 14U

This was noticed when compiling lily58/rev1 - I had a bad pin 7 and tried to use pin 14 (C7)

    "matrix_pins": {
        // left
        "cols": ["F6", "F7", "B1", "B3", "B2", "B6"],
        "rows": ["C6", "D7", "E6", "B4", "B5"]
    },
    "diode_direction": "COL2ROW",
    "split": {
        "enabled": true,
        "matrix_pins": {
            "right": {
                "cols": ["F6", "F7", "B1", "B3", "B2", "B6"],
                "rows": ["C6", "D7", "C7", "B4", "B5"]
            }},

This now compiles after adding to the converter.

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

daskygit, 2025-07-05:
A promicro doesn't have a buttom row, an elite c does. The pins for an elite-c are defined [here](https://github.com/qmk/qmk_firmware/blob/master/platforms/chibios/converters/elite_c_to_rp2040_ce/_pin_defs.h).

This isn't a bug and is working as intended.

closed 2025-07-05 by daskygit

rossw42, 2025-07-05:
the lily58/rev1 - has     "processor": "atmega32u4",
and im doing  -e CONVERT_TO=rp2040_ce 
- which is using the "promicro_to_rp2040_ce converter" / pin defs
so im not sure how to tell "qmk compile" to use the elite_c_to_rp2040_ce converter?

rossw42, 2025-07-05:
NM - found in the docs
PIN_COMPATIBLE = elite_c in rules.mk