qmk/qmk_firmware #25519: QMK instantly rewakes computer after suspend unless NO_USB_STARTUP_CHECK defined

techy-robot, 2025-07-27:
### Issue Description

I played with all the USB config settings yet my computer always rewakes a second after suspending, and it definitely is the keyboard. I've already tried `USB_WAIT_FOR_ENUMERATION = yes`, `USB_SUSPEND_WAKEUP_DELAY = 200` in `rules.mk`. Only `NO_USB_STARTUP_CHECK = yes` works, but that of course ignores keypress wake ups.

I also have added this to my config.h
```
#define RGB_MATRIX_SLEEP
#define RGB_MATRIX_TIMEOUT 300000
```
This is either a bug or I am missing some config setting causing the keyboard to keep my computer up. Nobody else has this issue, it seems to be the reverse of all the other issues. What am I missing?

My custom keyboard: https://github.com/techy-robot/JoyKey

drashna, 2025-09-20:
`USB_SUSPEND_WAKEUP_DELAY` isn't a valid rules.mk configuration. This is a setting that would go in config.h, or in json. 

As for re-waking after sleep, that's definitely weird.  Though, looking at your code, this is most likely due to do the custom encoder implementation, and is treating them as matrix activity and triggering the wakeup of the board

techy-robot, 2025-09-20:
Thank you, that makes much more sense now with my encoder on the matrix. I assumed the rewake routine would only happen on state change, not when a key is held. I did not know about the matrix mask, that looks very useful!

closed 2025-09-20 by techy-robot

mhalano, 2025-09-20:
@sigprof How can I check if my keyboard has one of these things, encoders or DIP switches? I have another keyboard, an X-Bows Nature, where happens the same and I'm trying to find out why but I don't want to bother anyone opening a new issue about it yet. Here is my configuration: https://github.com/mhalano/qmk_firmware/tree/mhalano/keyboards/xbows/nature/keymaps/mhalano-work If you want, I can open a new issue to discuss there, but I would like to check this encoders/DIP switches first.