qmk/qmk_firmware #25549: [Feature Request] QMK_GIT_HASH for userspace

szucsz, 2025-08-09:
### Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

### Description

I'd be interested to understand what is the idiomatic way to get the QMK_GIT_HASH but for the userspace repository, instead of `qmk_firmware`.

Background: I'd like to display the layout version on the display of the keyboard, rather than displaying the QMK Firmware version. Since there is no version attached to user-space, maybe the repo commit hash could be a useful build identifier or the git tag itself as well if that exists.

elpekenin, 2025-08-10:
Those macros are defined on a compile-time-generated file (`version.h`), you can see `lib/python/qmk/cli/generate/version_h.py` for details or even edit it. Best option is probably to send a PR that adds the userspace's hash to this file.

In the meantime, if you dont want to add "noise" to your fork, you can implement it at user-level (ie: no changes needed on `qmk_firmware`). As an example for you to reference, I have a [community module](https://github.com/elpekenin/qmk_modules/blob/main/version/rules.mk) that provides `QMK_MAJOR`, `QMK_MINOR` and `QMK_PATCH` (eg: `0`, `20`, `1`) -- based on a tzarc's snippet on discord

It boils down to creating a new makefile "step" (`$(GENERATED_VERSIONS_FILE): `) that performs the actual logic (writing the 3 bits of information to a file), and it is itself added as a dependency of the "magic" step `generate-files` in order to get run

You can then, optionally, also add it to the `POST_CONFIG_H` "list" of files, which are implicitly `#include`'d to every compilation... This way you don't even need `#include "my_file.h"` on your code

szucsz, 2025-08-15:
Thank you @elpekenin, I was able to get the userspace version. Might consider a PR later.

Let me just leave it here for future readers: the `POST_CONFIG_H +=` adds the target from the module's `rules.mk` to the dependencies. If I don't add the target to `POST_CONFIG_H`, the actions to generate the target won't run. In your case, I'd expect that after a `qmk clean` the `my_file.h` will be missing on the filesystem and the `#include "my_file.h"` will fail.

closed 2025-08-15 by szucsz