qmk/qmk_firmware #25642: [Bug] LAYOUT_split_3x6_2 missing in layouts/community

peterjc, 2025-09-09:
### Describe the Bug

Searching the repo for "LAYOUT_split_3x6_2" there are matches in the scottowing keyboard.

However https://github.com/qmk/qmk_firmware/tree/master/layouts/community only has `split_3x5_2`, `split_3x5_3`, and `split_3x6_3` listed.

There should be a `split_3x6_2` entry under https://github.com/qmk/qmk_firmware/tree/master/layouts/community for this 40-key variant with only two thumb-keys per hand.

### qmk doctor Output

```text

```

### Is AutoHotKey / Karabiner installed

- [ ] AutoHotKey (Windows)
- [ ] Karabiner (macOS)

zvecr, 2025-09-09:
Not every layout needs a corresponding community layout.

The fact there is only a single board using `LAYOUT_split_3x6_2` means its not worth introducing one.

peterjc, 2025-09-09:
Only one in the offical repository, but surely more outside it?

zvecr, 2025-09-09:
Can you provide any links of those that do?

mhdolsen, 2025-09-13:
@peterjc Community layouts have been removed from the repository. What is the value of adding an empty `LAYOUT_split_3x6_2` folder?